iNavFlight/inav #1332: Feature

handsomejackuk, 2017-02-25:
serialpassthrough on naze32 rev 5 targets is this possible ? betaflight has it

and if it is possible is serial passthrough available through softserial ports ?

with the new mwosd chrome configurator i found that communication through serial passthrough and through a bluetooth transmitter was possible so configuration of minim osd is real easy, and osd does not need to be unplugged to  modify osd configuration...

sppnk, 2017-03-05:
I like the idea

metalnow, 2017-03-06:
I will make a pr for this one.

digitalentity, 2017-03-06:
@metalnow thanks

hali9, 2017-04-01:
There is a gpspassthrough, so work only to GPS.
Unify it to all ports is great idea.

closed 2017-04-09 by digitalentity

RealNightEagle, 2018-01-03:
Hi I am new to the forum so sorry for posting in wrong area, if I have.
Can anyone tell me if I can serialpassthrough in INAV to connect to minim OSD via MWOSD software GUI?
I tried serialpassthrough 0 115200  It did not work
My minim OSD is on UART 1 on a SP F4 EVO
Cheers

davidngrc, 2018-08-17:
@RealNightEagle  I own a SP F4 EVO too, and try to do the osd pass through like you did.
here is how to do it. (I am in iNav 2.0RC6, MWOSD is in 1.6, osd is connect to UART 1)

1) in iNav Configurator -> CLI:    serialpassthrough 0 115200 rxtx
2) now, click the disconnect button on the top right corner (iNav Configurator).
3) launch chrome app MWOSD 1.6 free, download here https://github.com/ShikOfTheRa/scarab-osd-gui/tree/master/MW_OSD_GUI/R1.6
4) set the port same as the one in iNav Configurator, and set the baud rate to the one you set in CLI (in this case 115200), then click "Connect"
5) in MWOSD chrome app, it should show "Connected"
6) after you change the setting in MWOSD, save, and click "Disconnect"
7) unplug the fc battery and usb, and plugin again. you can now use the iNav Configurator again.