iNavFlight/inav #1335: Blackbox logs shows wrong Hz?

tomidland, 2017-02-26:
[image]
[image]

Here are two logs from the same quad flown just minutes apart. (Just hovering) First log is with 4kHz FC looptime, second is with 500Hz FC looptime. But they show very different noise frequencies in the BlackBox Explorer. They both claim to show from 0 - 500Hz. But one (or both) of them have to be wrong. In order to use the D-term and gyro notch filters properly one needs to know what frequencies to use..

In Betaflight the range changes depending on the FC loop. So if one has 1kHz FC loop, it shows 0-500Hz. If one has 2kHz FC loop it shows 0-1kHz and so on. (I assume this is because of the Nyquist frequencies sampling) Should not iNav do this too?

4kHz log : [LOG00001.TXT](https://github.com/iNavFlight/inav/files/801847/LOG00001.TXT)
500Hz log : [LOG00002.TXT](https://github.com/iNavFlight/inav/files/801849/LOG00002.TXT)

Drone : Quad 700
FC : Omnibus F4 v2 Pro (onboard sd)

DzikuVx, 2017-02-26:
I will try to take a look a this

DzikuVx, 2017-02-26:
OK, looks like frequencies are computed correctly, only spectrum analyzer range is broken and there are few reasons for it. Looks like it has to be fixed in few places

tomidland, 2017-02-26:
Oki

DzikuVx, 2017-02-26:
@tomidland issue (and some more) should be fixed in https://github.com/iNavFlight/blackbox-log-viewer/commit/db3b33b7640164fd54935d2caef43b82d52c9623

Please test and in case on any problems create a ticket in Blackbox Explorer project

closed 2017-02-26 by DzikuVx

tomidland, 2017-02-26:
Great! :) Will check it out!