iNavFlight/inav #1336: blackbox sd card

killerm1, 2017-02-26:
Is it possible to add a download log when using an sd card like when it has a datachip? or maybe make it have a storage device like a card reader? I have to dismantle my quad to get to the sd card to remove it so I can access the logs. im using the omnibusf4sd.

oleost, 2017-02-26:
No currently it's not possible.

Hopefully one day, as I also have to dig it out of an airplane everytime.

johnnyruz, 2017-04-05:
I would love this feature as well. I just built a pod style frame and it has a cutout for the USB port, but have to remove the pod to get to the SD Card.

krzysztofmatula, 2017-04-06:
The log downloading over serial/MSP is so terribly slow, so it may be completely unacceptable for SD card logs, where lower dividers are typically used...

I think we should aim for migrating to WiFi from all those USB/BT connections... :)
I'd love to access my UAV at high speed without actually connecting anything.

JulianGoesPro, 2017-06-04:
any news on this? I don't might letting the macbook run over night to download a log :D

flyfisher2, 2018-04-30:
Some form of downloading would be great, even if it took a long time
Apart from the difficulty of getting access to the reader to remove the card, I don't think that constantly inserting and removing the card is doing anything for the long term reliability of the FC.

mattschinkel, 2018-05-10:
I agree

Fast-Guy, 2018-06-16:
Wasn't this just added to Betaflight 3.4? It would be a great enhancement.

flyfisher2, 2018-06-16:
It surely would.

fiam, 2018-06-17:
I tried porting it from BF but and it worked fine for small files, but bigger ones got corrupted (tried it with a ~100MiB file). I won't have time to fix it for 2.0, but if anyone wants to take a crack at it the code is in the agh_msc branch.

flyfisher2, 2018-08-30:
This is only stale because nothing has happened.
It is still a real and important feature for installations where the SD card slot is not accessible without tearing the model apart.

mattschinkel, 2018-08-30:
I agree... Please don't close the thread.

flyfisher2, 2018-09-12:
Ditto

teckel12, 2018-09-13:
Is this possible by using the Betaflight configurator? Or won't it allow you to download the SD card contents with INAV firmware?

stronnag, 2018-09-13:
There is an experimental iNav branch, unfortunately while it works fine for small files, it corrupts big files. 
@teckel12 it requires the firmware to enable USB MSC mode. Then you just (auto)mount the FC as a mass-storage device.

stronnag, 2019-02-01:
> @stronnag : you probably mean https://github.com/iNavFlight/inav/tree/agh_msc? Can you bring this up-to-date, so we can try and see if we individually can find it useful?

Indeed I did. 
I can bring it up to date .... seems to "work" in that it mounts (as before). I shall see if it still corrupts files > 10MB.

stronnag, 2019-02-01:
Both Alberto and I experienced mismatched MD5sums (or similar checksums) against the initial commit. So far, on 2.1 / MATEKF405 I have success with 12MB and 24MB sizes, some bigger files under test. Just playing an MP3 is not a good test,I expect  they are robust (to one's ears) against minor corruption.

stronnag, 2019-02-01:
Further testing (concatenated BB log files)
72MB OK
144MB corrupt (md5sum mismatch)

putimir, 2019-02-01:
Ok, that's why I asked - I will do the MD5 hash comparisons and see - do the files need to be "live" or can be generated with like fsutil?

stronnag, 2019-02-01:
I don't think it matters what the files are. I copy them from the file system to the SDcard using a card reader, get the md5sum. Then insert into the fc, `msc`, rsync back to file system via msc and compare.

putimir, 2019-02-01:
Damn :(

```
C:\>dir d:\Temp\ue\orig\

Directory of d:\Temp\ue\orig
01. 02. 2019  17:07        12.582.912 12M.file
01. 02. 2019  17:08        25.165.824 24M.file
01. 02. 2019  17:08        52.428.800 50M.file
01. 02. 2019  17:06         5.242.880 5M.file
               4 File(s)     95.420.416 bytes
               2 Dir(s)  53.911.638.016 bytes free

C:\>dir d:\Temp\ue\msc\

Directory of d:\Temp\ue\msc

01. 02. 2019  17:07        12.582.912 12M.file
01. 02. 2019  17:08        25.165.824 24M.file
01. 02. 2019  17:08        52.428.800 50M.file
01. 02. 2019  17:06         5.242.880 5M.file
               4 File(s)     95.420.416 bytes
               2 Dir(s)  53.911.638.016 bytes free

C:\Users\jure>d:\Util\Unixutils\md5sum.exe d:\Temp\ue\orig\*

\efeebdda98ec1d7fb2ad83d23f0713bf *d:\\Temp\\ue\\orig\\12M.file
\77377273b0a4b61febdbf7bbf52b9db9 *d:\\Temp\\ue\\orig\\24M.file
\25e317773f308e446cc84c503a6d1f85 *d:\\Temp\\ue\\orig\\50M.file
\5f363e0e58a95f06cbe9bbc662c5dfb6 *d:\\Temp\\ue\\orig\\5M.file

C:\Users\jure>d:\Util\Unixutils\md5sum.exe d:\Temp\ue\msc\*

\d07fb92d6c252fbc08bcb62020a415db *d:\\Temp\\ue\\msc\\12M.file
\51959a93433931b3de1c6cfd35f15e61 *d:\\Temp\\ue\\msc\\24M.file
\e4d3a97beb5d7ffece91d5ed628c342a *d:\\Temp\\ue\\msc\\50M.file
\900a71b6448ee22c2c9bb27bb2f09473 *d:\\Temp\\ue\\msc\\5M.file
```

stronnag, 2019-02-01:
My final test resulted in a completely corrupt file, seemingly every block. I wonder if how far into the disk the file occurs is also a factor. Multiple I/O errors in syslog:
```
[72662.956784] sd 9:0:0:0: [sde] tag#0 FAILED Result: hostbyte=DID_ERROR driverbyte=DRIVER_OK
[72662.956791] sd 9:0:0:0: [sde] tag#0 CDB: Read(10) 28 00 01 e1 f7 88 00 00 38 00
[72662.956795] print_req_error: I/O error, dev sde, sector 31586184
[72662.956887] sd 9:0:0:0: [sde] tag#0 FAILED Result: hostbyte=DID_ERROR driverbyte=DRIVER_OK
[72662.956892] sd 9:0:0:0: [sde] tag#0 CDB: Read(10) 28 00 01 e1 f7 c8 00 00 28 00
[72662.956894] print_req_error: I/O error, dev sde, sector 31586248
[72662.957578] usb 2-3.2: USB disconnect, device number 26
[72662.962193] scsi 9:0:0:0: rejecting I/O to dead device
[72662.962201] print_req_error: I/O error, dev sde, sector 31586184
[72662.962207] Buffer I/O error on dev sde, logical block 3948273, async page read
[72662.962325] Buffer I/O error on dev sde, logical block 512, async page read
```

wx4cb, 2019-02-01:
@stronnag seems it might have issues with the data rate. transferring in smaller chunks might be better. definately looks like a buffer overrun or underrun.

Only other thing it might be is partition size? I know i have better luck with 2GB cards as opposed to a 16 or 32 gig card at least in the f405 wing

stephenaa, 2019-02-17:
Is transferring over MSP absolutely a no-go?
I am a relatively new iNav user, with a FC mounted in a way that makes accessing the SD card very hard. In my non-flying time I am a SW engineer, and eager to help find a solution to this.

putimir, 2019-03-01:
Yea, any progress perhaps. Especially with the work done on the inav dashware profile :)

Seeelefant, 2019-03-22:
In PX4 and ArduCopter this feature is available, maybe it´s worth to look at them.

stephenaa, 2019-03-30:
Does PX2 and Arducopter use the MSC method? Betaflight uses that now, right?
I would like to know why going through all the extras of MSC and FAT makes this approach better (no CRC errors) than retrieving data directly through MSP. Unless of course the issue is with the receiving end (configurator). 

I would like to put some time into this - so every comment and input is appreciated.

shellixyz, 2019-03-30:
No they don't. PX4 and Ardupilot transfer the logs through Mavlink. Betaflight yes supports MSC. Implementing MSC makes logs management more natural, it is then possible to use existing OS file managers and we don't need anything special in the configurator to manage files. No need to reinvent the wheel.

stephenaa, 2019-03-31:
OK, I agree. I will take a look. I need to target "development" for a potential PR, not "master" - right?
I will apply the agh_msc patch on top of the "development" branch.

giacomo892, 2019-03-31:
@shellixyz we have log download capability via MSP on targets with flash storage. Would be nice implementing (exteding the current interface) something similar for the sdcard. 
We can even potentially remove the FAT layer and just write the data on the SD as we need. 
Could be nice that MSP provides the list of the logs and you can download what you want on the fly via USB with the help of the configurator.
@stephenaa yeah. development branch

Seeelefant, 2019-03-31:
Another cool feature of PX4 is: You can download the log via WiFi, no fiddling around with Micro USB plugs etc....highly appreciated feature :-)

shellixyz, 2019-03-31:
I'm against removing the FAT layer on the SDcard but yeah we could have the possibility to download logs from the configurator, I'll let you do it @giacomo892 :wink:  .The best would be to have both options so that as @Seeelefant pointed out it could be possible to download logs through other means than USB.

stephenaa, 2019-03-31:
I think it makes most sense from the user(s) if the logs are available directly from the SDcard if connected to a computer, and through the configurator when the card is still inserted in the FC. If MSP is used for log transfer it should be relatively simple to do that over WiFi for boards/setups with WiFi HW.
@giacomo892 : Yeah, wouldn't a unified access to log - no matter the storage medium be the most "natural" ?

krzysztofmatula, 2019-03-31:
Definitely possibility to download logs without removing card is a welcome feature.
But doing so over MSP would likely fail for MAC users unless the serial port issue is finally somehow fixed...

stephenaa, 2019-03-31:
@krzysztofmatula Being a MAC user I am very interested to make it work :-)
Is there any proper reason for the dataflash reading not to also encapsulate reading from SD card? It seems unnessesary to expose the different storage methods to the user?

krzysztofmatula, 2019-03-31:
There is no particular reason I think :)

The difference is that SD logs are organised into files, while flash chip logs are downloaded as a whole.
Best if it also could get unified, allowing to split the flash log into proper files.

One drawback of using MSP is that the protocol is slow and serial link is also is slow. Having quite big SD cards many use quite hight logging rate, which would take ages to download over MSP.

stephenaa, 2019-03-31:
@krzysztofmatula Maybe a proper compression algorithm will be able to fix the speed issue? The nature of the data should make efficient compression possible.

I am considering modelling a new interface (MSP2 cads), that allows a file-oriented storage of logs. For now that can also cover dataflash as a special case having only one file. At a later point dataflash could be extended to include multiple files - which then just need the adaptation to the file-oriented MSP interface.

krzysztofmatula, 2019-03-31:
Years ago there was PR for compression (never merged) - see #1253.
See also some discussion here: #1640

stephenaa, 2019-03-31:
OK, I will start with the basic "file" access and all the protocol/transfer code. Then optimizing the transfer may be a next step. Going this way should make some future MSP over WiFi also support SD card log retrieval - while still keeping FAT files on SD card AND allowing for dataflash to be handled the same way.

stephenaa, 2019-04-27:
I have started on this, trying to do a simple file oriented interface that will serve the log files over MSP. @krzysztofmatula What are the details of the mac serial issues you mentioned earlier?

L4ky, 2019-09-09:
Any chance to see this feature implemented in near future?

stephenaa, 2019-09-09:
My work on it stalled over the summer. But I still would like to have the feature. 
Daytime job took my cycles however...

Sendt fra min iPhone

stronnag, 2019-12-18:
There is an updated, experimental branch / PR #5277 that appear to be working MSC (mass storage device class), meaning you can mount the FC (SD card) as an OS file system via USB to read BB logs (and delete them).

## Testing

So far MSC mode has been tested on a MATEKF405, with iNav created log files of sizes from 8400 bytes to 109MB, as follows:

* Use extant log files, or have the FC create log files (at 1000 looptime, 1:2 bbl rates, 109MB takes around 50 minutes).
* After disarming the FC, enter MSC mode (see below)
* Once the host OS has mounted the FC (SD card), copy the BBL file(s) to a temporary directory
* Dismount (eject) the SD card
* Remove the SD card and read it directly (e.g. via a card reader), copy the same fle(s) to another temporary directory.
* For each file, verify that the file copied over MSC is the same as the file copied directly (e.g. using a card reader). On Linux, you can use `cmp` to compare binary files, or use `md5sum` on each file.
* The files should be identical.

For all of my 35 test files, sizes ranging from 8400 bytes to 109MB, the files copied by MSC and directly are identical (tested on Arch Linux).

Delete is also tested as working. Writing via MSC is not tested (or really relevant).

The PR has also been successfully tested by rissah (RCG) with MATEKF722 and Windows 10.

## Performance

Reading is quite slow, typically c. 340kBs, for example:

```
####################
## Using MSC mode ##
####################
# FC is automounted to /run/media/jrh/BBOX-QUAD by the OS
$ rsync -P /run/media/jrh/BBOX-QUAD/LOGS/LOG00035.TXT /tmp/msclogs/
LOG00035.TXT
     55,856,827 100%  339.15kB/s    0:02:40 (xfr#1, to-chk=0/1)
```

```
#########################
## Using a card-reader ##
#########################
# SD Card is automounted to /run/media/jrh/BBOX-QUAD by the OS
$ rsync -P /run/media/jrh/BBOX-QUAD/LOGS/LOG00035.TXT /tmp/sdclogs/
LOG00035.TXT
     55,856,827 100%   19.26MB/s    0:00:02 (xfr#1, to-chk=0/1)
```
i.e c. 2.5 seconds for the card reader, 2 minutes 40 seconds for MSC (60 times slower). However, if the card is relatively inaccessible, this is a reasonable trade-off

## Comparison and Integrity

```
$ cmp /tmp/{msc,sdc}logs/LOG00035.TXT
# no differences reported ...
```

```
$ md5sum /tmp/{msc,sdc}logs/LOG00035.TXT
7cd259777ba4f29ecbde2f76882b1840  /tmp/msclogs/LOG00035.TXT
7cd259777ba4f29ecbde2f76882b1840  /tmp/sdclogs/LOG00035.TXT
```
You should also be able to run blackbox utilities (e.g. the iNav specific `blackbox_decode`) without errors on the files, e.g.

```
$ blackbox_decode --stdout --merge-gps > /dev/null /tmp/msclogs/LOG00035.TXT
Log 1 of 1, start 36:00.888, end 62:00.851, duration 25:59.963

Statistics
Looptime           1006 avg            9.2 std dev (0.9%)
I frames   48405  104.6 bytes avg  5062215 bytes total
P frames  726064   69.2 bytes avg 50246994 bytes total
H frames     380   10.0 bytes avg     3800 bytes total
G frames   15674   21.4 bytes avg   334701 bytes total
S frames    6198   33.0 bytes avg   204534 bytes total
Frames    774469   71.4 bytes avg 55309209 bytes total
Data rate  496Hz  35806 bytes/s     358100 baud

3 frames failed to decode, rendering 4 loop iterations unreadable. 4 iterations are missing in total (4ms, 0.00%)
774472 loop iterations weren't logged because of your blackbox_rate settings (779980ms, 50.00%)
```

## Usage

To put the FC in MSC mode:

* Enter the CLI
* Enter the CLI command `msc` ; the FC will reboot
* Close the CLI tool (`cliterm`, configurator etc.)
* Wait for the device to be recognised as USB storage device by the operating system (may take some time, 10-15 seconds perhaps).
* Copy files off the MSC mounted FC (sd card) (`cp`, file manager)
* Dismount / eject the FC (sd card) card using the standard OS method
* Power-cycle the FC to exit MSC mode.

## Background

This has been a janitorial task, there are **no** changes of any substance over the original work by the βF developers and @fiam. In the interim, there have been minor updates to the USB subsystem which may have helped, and the testing methodology has been somewhat rationalised.

Additional testing would be welcomed ....

stronnag, 2019-12-18:
hex files for F4/F7 SD card targets (E&OE)
http://jrh.sdf-eu.org/inav-msc/

putimir, 2019-12-18:
Hi @stronnag, using Matek F405-WING, Kingston SDHC/U1/ Class10 /16GB , Windows 10/64, I have done 3 rounds of:
1. powering on the FC through USB and entering msc through cli
2. copying a ~90MB file to the mounted drive
3. unmount drive via "eject..."
4. disconnect FC from USB
5. again powering on the FC through USB and entering msc through cli
6. downloading the previously uploaded file

I experianced no errors and all 3 times the file was identical: 

c:\Temp>d:\Util\Unixutils\md5sum.exe JESC-Configurator_win64_1.2.6.orig.zip
1f75c3f5fb7a238fa312067e12bb47de *JESC-Configurator_win64_1.2.6.orig.zip

c:\Temp>d:\Util\Unixutils\md5sum.exe JESC-Configurator_win64_1.2.6.msc.zip
1f75c3f5fb7a238fa312067e12bb47de *JESC-Configurator_win64_1.2.6.msc.zip

So is the build *relatively safe* to fly? I'd do some flying with other FCs to see it *in action*.

Is the:

`set blackbox_rate_num = 1
set blackbox_rate_denom = 2`

in the CLI enough to start logging? The 2.3.2 configurator does not work with this build, and strangely, the 2.3.0/MATEKF405SE gives me this on th eBlackbox tab:

[image]

stronnag, 2019-12-18:
You need `feature BLACKBOX` if it's not default.
There are builds of the dev configurator at http://seyrsnys.myzen.co.uk/inav-configurator-linux/ (not just linux), which recognises 2.4.0 as a valid version; apologies, forgot to mention that before.
It should be safe to fly, I flew 2.4 development build on Monday.

And thanks for testing.

putimir, 2019-12-18:
Ok, yea, I already figured it out :) So I'll try to fly a.s.a.p. Is there any need to try different boards? I have  MATEKF405SE, MATEKF722SE, MATEKF722...

stronnag, 2019-12-18:
I think we've established that it's working on both F4 and F7. 

A test with MacOS would be helpful (someone); we've done Linux, Win7 and Win10.

hali9, 2019-12-20:
Tested under Win7 (32bit) and Android with OTG and work well.
OMNIBUSF4PRO with 1GB noname SD.
After putting CLI msc disc appear in 5 seconds. Transfer 380KB/s. 
Can delete log and freespace file, so after rebot FC freespace have again 1GB size.
Realy great.

stronnag, 2019-12-20:
> Tested under Win7 (32bit) and Android with OTG and work well.
> OMNIBUSF4PRO with 1GB noname SD.
> After putting CLI msc disc appear in 5 seconds. Transfer 380KB/s.
> Can delete log and freespace file, so after rebot FC freespace have again 1GB size.
> Realy great.

Thanks for testing. I should fix the final build issues (for non-MSC targets) today.

stronnag, 2019-12-20:
> Thanks for testing. I should fix the final build issues (for non-MSC targets) today.

Oh, this just gets better. It works with `ONBOARDFLASH` targets as well ... at least once the build is updated.

stronnag, 2019-12-20:
Now (not yet pushed to GH), it reads BBLs from internal flash as well using USB MSC. Nice, very nice indeed.

However, that (virtual) file system has to be read-only and clearing the internal flash will have to be done via the configurator, the CLI or other external application.

stronnag, 2019-12-21:
Should (really) be done ... 
hex files for F4/F7 **SD card** and **Internal Flash** targets (E&OE), gzip compressed: http://jrh.sdf-eu.org/inav-msc/

'development' (pre-2.4) configurator if required: http://seyrsnys.myzen.co.uk/inav-configurator-linux/

Documentation: https://github.com/iNavFlight/inav/blob/agh_msc_revisited/docs/USB_Mass_Storage_(MSC)_mode.md

stronnag, 2019-12-22:
As #5277 has been merged into development, this can be closed.

closed 2019-12-22 by stronnag