iNavFlight/inav #1343: Feature request: FPV angle mix mode from betaflight.

oleost, 2017-02-28:
Cant fint the original PR, but found some after some fixes.

https://github.com/betaflight/betaflight/pull/1118
https://github.com/betaflight/betaflight/pull/171
https://github.com/betaflight/betaflight/pull/2117
https://github.com/betaflight/betaflight/pull/2202

digitalentity, 2017-03-01:
Not sure this mix is a good thing, but if it's wanted - we'll add it. 
We just need to make sure it doesn't mess up with navigation modes.

oleost, 2017-03-01:
For me personally the mix could simply get disabled as soon as any navigation mode is active.

DzikuVx, 2017-03-01:
As long as this new mix can be disabled by user, it's fine for me

oleost, 2017-03-01:
It`s an new "Flight mode" thats called `FPV ANGLE MIX`

digitalentity, 2017-03-01:
Do we have spare bits in `boxId_e`?

DzikuVx, 2017-03-01:
BOXLLIGHTS is not used ATM. I want to iplement that some day but we can
reuse that.
Also, there are 28 boxes right now, so we still have few

DzikuVx, 2017-03-01:
And that might be a good moment to finish #469 and bump `rcModeActivationMask` to 64 bits as well

DzikuVx, 2018-05-09:
I belive this is done in #3037

closed 2018-05-09 by DzikuVx