iNavFlight/inav #1345: Galileo GNSS configuration missing

Schalonsus, 2017-03-01:
Referring to this https://www.rcgroups.com/forums/showpost.php?p=36984859&postcount=11720
I looked in the code but can not find anything configuring Galileo
https://github.com/iNavFlight/inav/blob/master/src/main/io/gps_ublox.c

Do i overlook it or is just not implemented?

digitalentity, 2017-03-02:
Do we need to explicitly enable GALILEO on Ublox modules?

Schalonsus, 2017-03-02:
I think on default configuration it is off. But i dont have a galileo able gps atm to confirm this.
And to set this in u-center is not the solution i think. If we configure gps with iNav then it should be all options we need.

oleost, 2017-03-02:
@sambas  has experience with Galileo.

giacomo892, 2017-03-26:
I've notice that from my area I can receive BeiDou. 

Enabling  GPS+GLONASS+BeiDou together should be feasible without any firmware upgrade on most M8N gps giving some more sats to lock.

[image]

Sat 212 in the pic is from BeiDou constellation.

fiam, 2017-10-25:
Just tested a couple of M8N receivers, one from HGRLC and another one from... who knows. Both of them support Galileo, but neither had it enabled by default.

I was able to permanently change the setting by opening U-Center, enabling Galileo and saving to flash. I will flight test this in the weekend and see if it makes any measurable difference.

In the meantime, I think if this turns out to be beneficial we should either enable it by default when possible or introduce a setting to enable it. According to the data sheet, changing the GNSS configuration to an unsupported value will just reject the change, leaving the module in its previous configuration, so it shouldn't cause any problems.

DzikuVx, 2018-02-05:
I belive we have this thanks to @stronnag and #2583

closed 2018-02-05 by DzikuVx