iNavFlight/inav #1346: Feature request: Softserial on Omnibus targets

isilcala, 2017-03-01:
Latest version of Betaflight does enabled softserial on Omnibus targets, perhaps we could port it to iNav?

raul-ortega, 2017-03-02:
This is necessary, because like in my case, I have UART1 for MSP, UART2 for GPS and UART3 for RX (FrSky receiver), and I need an additional port to send telemetry to ground station :(

scoop3, 2017-03-10:
Agreed! I'm on an OmnibusF4Pro, and I need ONE MORE port for Telemetry. Please!! 😄

Riffer, 2017-03-11:
Afaik Omnibus already support soft serial:
From https://github.com/iNavFlight/inav/blob/master/docs/Board - Omnibus F4.md

> 4 UART ports (UART1, UART3, UART6, Softserial1)

From https://github.com/iNavFlight/inav/blob/master/docs/Board - OMNIBUS.md
> All UARTs are Serial RX capable.

scoop3, 2017-03-11:
Sorry, yes it does, what I need is for iNav to incorporate resource remapping, so that softserial can be assigned to pwm5 (current BF supports)

scoop3, 2017-03-11:
Unless I'm missing something... Does anyone know what the default pin is for softserial 1 and 2 in iNav?

oleost, 2017-03-11:
https://github.com/iNavFlight/inav/blob/master/docs/Board%20-%20Omnibus%20F4.md

oleost, 2017-03-11:
Closing this one, feel free to reopen if anything is not working as expected.

closed 2017-03-11 by oleost

urlu75, 2017-04-01:
> Sorry, yes it does, what I need is for iNav to incorporate resource remapping, so that softserial can be assigned to pwm5 (current BF supports)

any tips/news on how to use PWM5 and 6 for softserial instead of small pad ch5 & 6 ?

scoop3, 2017-04-01:
I believe, currently, as resources aren't currently remappable, custom firmware is the only option. :-/

scoop3, 2017-04-01:
I agree tho, those pads are tiny.. and I already lifted ch6... So gonna have to figure out how to compile something new..

urlu75, 2017-04-01:
It should be somewhere in target.h I guess ...
#define USE_SOFTSERIAL1
#define SOFTSERIAL_1_TIMER_RX_HARDWARE 4 // CH5
#define SOFTSERIAL_1_TIMER_TX_HARDWARE 5 // CH6

I already made some test with different value, no success, compilation failed or it just doesn't work.
I don't what code lines have to be changes to move softserial to pin A1 and A8 (PW5 & 6).

scoop3, 2017-04-01:
You might need to unassigned motor 5 and 6 first somewhere as well, so it's not assigned twice?

scoop3, 2017-04-01:
I'd they'd just incorporate the remappable assets from bf3.1, that'd be perfect haha

stronnag, 2017-04-01:
I somehow doubt that; it would just break the diverse platform types that work on iNav and do not work in βF.

scoop3, 2017-04-01:
If*

How would being able to remap resources to any pin of your choosing do anything other than add diversity to iNav?

stronnag, 2017-04-01:
So generate the PR and the documentation that does so; the latter being something missing in βF.
For most users this is so far not practical.

Riffer, 2017-04-02:
Is softserial really to be used by those two incredible tiny pads on the bottom side of the controller?

scoop3, 2017-04-02:
Haha yes... I already lifted my ch6 pad too :(

Perfect example of why remappable resources would be great.... :-/

urlu75, 2017-04-02:
for sure with time pad 5 and 6 will be damage and softserial is needed for telemetry :/
GPS, compass and receiver are using other UART.
Remappable ressources or some tips to build a custom firmware with softserial on pwm5 & 6...

Gavsv8, 2017-05-15:
I followed the doc for the ch5/6 pads and softserial but had to set the baud rate to 19200 on my Flip32 F4 pro 2.5 with XSR to get it working. There was a note somewhere else about this being the maximum for softserial.

TwoToneEddy, 2017-09-01:
CH5 and CH6 are no longer available at all on the V4 board. If you're using SBUS, external mag and GPS you can't use telemetry which is a pain if you want to do waypoint flying. Can we remap soft serial to pwm5 and 6? I'm just looking into it now but I'm not sure exactly how to do it.

cocolimon, 2017-09-12:
Where is the ch5 and ch6?

teckel12, 2017-10-11:
@cocolimon You'll probably find this image useful:

[image]

Note that with INAV v1.7.4 the LED pin will be on the J1 connector.  With 1.7.3 it's on PWM5 as illustrated in the above image.

haplm, 2017-11-30:
I need to second @TwoToneEddy - I have v5 board, and CH5/6 is not there either. Is it possible to move this to PWM5 and PWM6, and have 2 softserials, as in Betaflight?

TwoToneEddy, 2017-12-01:
I haven't got one of these boards anymore but can't you just use resource reassignment on the cli? I never tried it.

haplm, 2017-12-01:
If I understand #1145 correctly, resource reassignment is not available in iNav, and maybe will be added in 2.0 when it is released...

teckel12, 2017-12-01:
I don't believe resource reassignment (at least how betaflight does it) is on the schedule for INAV 2.0. So I wouldn't hold your breath for this.

zector555, 2017-12-14:
I have been having this issue too, my board does not have the 2 small pads, but PWM5/6 are available.  Im using inav as well.  I think i am close, but softserial still will not enable for me.  This is what ive done (basicly tried to change the ch pads to the pwm pads):

OMNIBUSF4/target.c
// original line { TIM5, IO_TAG(PA1), TIM_Channel_2, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM5, TIM_USE_MC_MOTOR | TIM_USE_MC_SERVO | TIM_USE_FW_SERVO }, // MOTOR_5
{ TIM5, IO_TAG(PA1), TIM_Channel_2, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM5, TIM_USE_ANY }, // MOTOR_5 mod to soft serial

// original line { TIM1, IO_TAG(PA8), TIM_Channel_1, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM1, TIM_USE_MC_MOTOR | TIM_USE_MC_SERVO | TIM_USE_FW_SERVO }, // MOTOR_6
{ TIM1, IO_TAG(PA8), TIM_Channel_1, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM1, TIM_USE_ANY }, // MOTOR_6 mod to soft serial

OMNIBUSF4/target.h
#define USE_SOFTSERIAL1
// original line #define SOFTSERIAL_1_RX_PIN PC8
#define SOFTSERIAL_1_RX_PIN PA1
// original line #define SOFTSERIAL_1_TX_PIN PC9
#define SOFTSERIAL_1_TX_PIN PA8

I got a response on FB inav group for this, someone said it may not be possible  because of the timers shared between gpio pins. Can anyone here weigh in?

teckel12, 2017-12-14:
@zector555 I believe the issue is the timers.  Timers can be challenging to deal with on microcontrollers. And most have limitations where timers oy work on certain pins. But, others have mentioned BF better able to reassign the pins, so that's where I would look first. I have a Pro, so I have the ch5/ch6 pads for SS.

zector555, 2017-12-14:
Timer issues is way above my level.  I was able to compile, and get the softerial showing in resources, however it does not appear to push any smart port data.  When i get more time, i was going to hook up a ftdi and see if any data is coming out.  Is it worth spending more time on, or am i just beating my head on a wall?

Raul has been helping me on the inav FB group too.

Thanks
# resource
IO:
----------------------
A00: FREE 
A01: SOFTSERIAL1 UART TX
A02: MOTOR4 OUT
A03: MOTOR3 OUT
A04: MPU CS
A05: SPI1 SCK
A06: SPI1 MISO
A07: SPI1 MOSI
A08: SOFTSERIAL1 UART RX

shellixyz, 2017-12-14:
I also tried to move softserial pins to M5/M6 in #2556 without success.

r0ba, 2017-12-28:
Whats about regular Omnibus board (f3)?