iNavFlight/inav #1347: Preset PID of Flying Wing Z84 are not correct

isilcala, 2017-03-01:
Preset value of P_PITCH & P_ROLL are both 2, should be 20 I suppose?

isilcala, 2017-03-01:
but for 600mm flying wing those Ps are 20?

digitalentity, 2017-03-01:
With PIFF controller there are different ways to achieve a good tune.

Schalonsus, 2017-03-01:
I can confirm for planes you should use low P value and therefore higher FF

isilcala, 2017-03-01:
Z84 and 600mm wings are pretty close things, if with PIFF controller, 2 is the right value for Z84, then for 600mm wings the default P should be 2 as well? it seems one of the preset value is incorrect (probably the 600mm wing?).

Schalonsus, 2017-03-01:
With my 1.2m wing i have P-Pitch = 7 and P-Roll = 17
But i looked at my BB log and saw that i need to reduce P-Pitch a little bit and P-Roll a lot.
So i will try P-Pitch = 5 and P-Roll = 8

oleost, 2017-03-01:
@isilcala As Digitalentity pointed out there are different ways to archieve a good tune with PIFF.

Yes 600mm and Z84 are close, but two different people have tuned theyr aircrafts.
One where P-gain does alot of work, and the other one where FF-gain does it.

I ( I made the Z84 preset ) prefer to have the FF-gain do most if it, as it will mimick more flying in passthrough but with added stabiliation.

isilcala, 2017-03-01:
Good to know those values are intentionally set and correct. I thought there might be a mistake to see such a big difference in values but good it's not the case. Very appreciate you guys created the PIFF controller for airplanes, can't wait to try it out!

closed 2017-03-01 by oleost