iNavFlight/inav #1348: Getting parse error when applying custom mix

oleost, 2017-03-01:
Verified on fresh install of INAV1.6 on both spracingf3mini and spracingf3.

Trying to example load:
mixer CUSTOMAIRPLANE
mmix reset
mmix 0 1.000 0.000 0.000 0.000
smix reset
smix 0 3 0 -100 0 0 100 0
smix 1 4 0 -100 0 0 100 0
smix 2 2 1 -100 0 0 100 0

Gives this:

INAV 1.6:
[image]

INAV 1.5.1:
[image]

Schalonsus, 2017-03-01:
Remove the last zero from smix, this is since  some versions i think.

oleost, 2017-03-01:
Is this done on purpose?

DzikuVx, 2017-03-01:
I remember a commit @digitalentity that removed trailing `0`

digitalentity, 2017-03-02:
Yes, we removed `SERVOx` modes for some time now as nobody seemed to be using them. `smix` `<box>` rule was a remainder of that and was removed in 1.6

Mixer docs indeed need updating.

closed 2017-03-02 by digitalentity