iNavFlight/inav #1354: Virtual current monitor report current x10

malaz80, 2017-03-04:
On spracingf3 the current detected by the virtual sensor is multiplied by 10, on standby should bring 320mA but instead shows 3.2A

digitalentity, 2017-03-04:
Virtual sensor is not a "turn-key" solution. You have to configure it's parameters properly using a real ammeter.

malaz80, 2017-03-04:
Ok 1.5.1 work fine

digitalentity, 2017-03-04:
Please upload your dump from 1.5.1

malaz80, 2017-03-04:
[Uploading inav_backup_3.1.2017.15.29_dump_151.txt…]()
[Uploading inav_backup_3.2.2017.11.42_dump_160.txt…]()

malaz80, 2017-03-04:
# dump

# version
# INAV/SPRACINGF3 1.5.1 Jan  5 2017 / 11:33:42 (556168d)
# pflags
# Persistent config flags: 0x00000000
# dump master

# mixer
mixer QUADX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -UNUSED_1
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -UNUSED_2
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature VBAT
feature RX_SERIAL
feature MOTOR_STOP
feature GPS
feature FAILSAFE
feature CURRENT_METER
feature LED_STRIP
feature BLACKBOX
feature PWM_OUTPUT_ENABLE

# beeper
beeper GYRO_CALIBRATED
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
beeper ACC_CALIBRATION
beeper ACC_CALIBRATION_FAIL
beeper READY_BEEP
beeper MULTI_BEEPS
[... 389 lines cut ...]
set acc_soft_lpf_hz = 15
set dterm_lpf_hz = 40
set yaw_lpf_hz = 30
set yaw_p_limit = 300
set iterm_ignore_threshold = 200
set yaw_iterm_ignore_threshold = 50
set rate_accel_limit_roll_pitch = 0
set rate_accel_limit_yaw = 10000

# dump rates

# rateprofile
rateprofile 0

set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 100
set thr_expo = 0
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set tpa_rate = 0
set tpa_breakpoint = 1500

#

malaz80, 2017-03-04:
# dump

# version
# INAV/SPRACINGF3 1.6.0 Feb 28 2017 / 13:44:32 (0def456)

# resources

# mixer
mixer QUADX

mmix reset

# servo
servo 0 1000 2000 1500 90 90 100 -1
servo 1 1000 2000 1500 90 90 100 -1
servo 2 1000 2000 1500 90 90 100 -1
servo 3 1000 2000 1500 90 90 100 -1
servo 4 1000 2000 1500 90 90 100 -1
servo 5 1000 2000 1500 90 90 100 -1
servo 6 1000 2000 1500 90 90 100 -1
servo 7 1000 2000 1500 90 90 100 -1

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature VBAT
feature RX_SERIAL
feature MOTOR_STOP
feature GPS
feature LED_STRIP
feature BLACKBOX
feature PWM_OUTPUT_ENABLE

# beeper
beeper GYRO_CALIBRATED
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
[... 395 lines cut ...]
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set tpa_rate = 0
set tpa_breakpoint = 1500

#

malaz80, 2017-03-04:
Entering CLI Mode, type 'exit' to return, or 'help'

# dump

# version
# INAV/SPRACINGF3 1.6.0 Feb 28 2017 / 13:44:32 (0def456)

# resources

# mixer
mixer QUADX

mmix reset

# servo
servo 0 1000 2000 1500 90 90 100 -1
servo 1 1000 2000 1500 90 90 100 -1
servo 2 1000 2000 1500 90 90 100 -1
servo 3 1000 2000 1500 90 90 100 -1
servo 4 1000 2000 1500 90 90 100 -1
servo 5 1000 2000 1500 90 90 100 -1
servo 6 1000 2000 1500 90 90 100 -1
servo 7 1000 2000 1500 90 90 100 -1

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature VBAT
feature RX_SERIAL
feature MOTOR_STOP
feature GPS
feature CURRENT_METER
feature LED_STRIP
feature BLACKBOX
feature PWM_OUTPUT_ENABLE

# beeper
beeper GYRO_CALIBRATED
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
[... 398 lines cut ...]
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set tpa_rate = 0
set tpa_breakpoint = 1500

#

digitalentity, 2017-03-04:
INAV-1.6. Your settings. Nothing wrong:
[image]

malaz80, 2017-03-04:
with set current_meter_offset = 3
but with set current_meter_offset = 32 (current_meter_offset = Imin * 100)

malaz80, 2017-03-04:
[image]

malaz80, 2017-03-04:
# dump

# version
# INAV/SPRACINGF3 1.6.0 Feb 28 2017 / 13:44:32 (0def456)

# resources

# mixer
mixer QUADX

mmix reset

# servo
servo 0 1000 2000 1500 90 90 100 -1
servo 1 1000 2000 1500 90 90 100 -1
servo 2 1000 2000 1500 90 90 100 -1
servo 3 1000 2000 1500 90 90 100 -1
servo 4 1000 2000 1500 90 90 100 -1
servo 5 1000 2000 1500 90 90 100 -1
servo 6 1000 2000 1500 90 90 100 -1
servo 7 1000 2000 1500 90 90 100 -1

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature VBAT
feature RX_SERIAL
feature MOTOR_STOP
feature GPS
feature CURRENT_METER
feature LED_STRIP
feature BLACKBOX
feature PWM_OUTPUT_ENABLE

# beeper
beeper GYRO_CALIBRATED
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
[... 396 lines cut ...]
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set tpa_rate = 0
set tpa_breakpoint = 1500

#

digitalentity, 2017-03-04:
Ok, confirmed

digitalentity, 2017-03-04:
Your dump for 1.5 has `set multiwii_current_meter_output = OFF` while 1.6 has `set multiwii_current_meter_output = ON` 

Change to OFF and you'll be fine

closed 2017-03-04 by digitalentity

malaz80, 2017-03-04:
yes, it works!!!