iNavFlight/inav #1355: Failsafe kill switch refactoring

digitalentity, 2017-03-04:
I wonder if failsafe kill switch setting should be removed completely. If you can spare a switch for FAILSAFE mode you surely can space a switch for ARMing which will give you the same manual disarm opportunity.

If you want actual signall loss do disarm the machine the best way to do that is to configure failsafe procedure to DROP the machine.

martinbudden, 2017-03-04:
I've always wondered why anyone would want to set failsafe with a switch. It is useful for testing failsafe mode, I suppose, but I can't really see a use for it in normal flight.

digitalentity, 2017-03-04:
`FAILSAFE` flight mode is useful for receivers that can't go to no-pulses and only have pre-set failsafe (like many Spektrum-compatible receivers).

I'm talking about `failsafe_kill_switch` setting which bypasses failsafe procedure and disarms instantly if `FAILSAFE` flight mode is activated from a switch (basically implementing a *disarm* switch).

Redshifft, 2017-03-04:
@martinbudden  Essentially It needs to be called just "kill switch" and do just that, a safety feature.
Also any links to the failsafe mechanism be  removed as far as possible

raul-ortega, 2017-03-06:
I would remove it!!!

digitalentity, 2017-03-06:
I suggest changing it to a flight mode called `KILL SWITCH` that would instantly disarm and prevent rearming.

oleost, 2017-03-06:
@digitalenitity do we need it? Why just not disarm?

Redshifft, 2017-03-06:
@digitalentity  Yup, pretty much what it does now with a less confusing name - perfect

sppnk, 2017-03-06:
I vote for remove enterely this 'kill switch' thing. As @digitalentity said, you can set failsafe to drop  the aircraft, so if this is causing troubles and  disarming machines in-flight it should be removed.

Dronek, 2017-03-06:
> I suggest changing it to a flight mode called  KILL SWITCH  that would instantly disarm and prevent rearming.

@digitalentity : thinking of fixed wing, it may make sense to additionally kill PWM output when killswitch is activated. As of now a disarmed plane will still direct roll/pitch commands to the servos. IMHO, a killswitch should completely kill any thrust/steering related output.

digitalentity, 2017-03-06:
@Dronek good point - and not only on airplanes, but also on multirotors. To let ESCs know that everything is dead.

oleost, 2017-03-06:
I feel we have two choices.

1. Keep it as it, but update documentation more.
2. Remove it entirely

Or third:

Remove it entirely. Create a new flight mode which is "Kill switch / output 0" and let it disable all output just like the pwn feature. It will not disarm only kill outputs.

oleost, 2017-03-28:
#1475

closed 2017-03-29 by digitalentity