iNavFlight/inav #1359: Althold mode at big tilt boosts throttle too much causing over controls.

RBirdie001, 2017-03-04:
When copter is in althold mode and is tilted significantly (I guess over 30 degrees), althold boosts throttle extremely up so that overcontrols to crazy dance. I seen it twice:
1) Tricopter with iNAV 1.5 in althold, manual control in horizon mode. I got to close to a tree branch so I had to use bigger input on roll stick (but didn't touch throttle stick). Copter got to crazy overshooting so I was lucky to land it. Video: https://www.youtube.com/watch?v=dP9BRAIkA5E
2) Quad with iNAV 1.6RC2, newly set nav_max_speed = 500. I flew normally in horizon for a minute, then I activated RTH (it means also althold). Quad started first turning towards me, but then started uncontrollable rolls. Video: https://www.youtube.com/watch?v=9JbhxLmhgyQ
More description at videos.
Can be maximal throttle correction by althold limited? 
Thanks!
Roman

digitalentity, 2017-03-05:
Please share 1) details of your setups; 2) configuration dumps; 3) blackbox logs if available.

My thoughts: 

 1. AltHold works perfectly with tilts up to 60 deg (tested). Your first video doesn't look too bad. If that was an uncontrolled climb it looks more like an accelerometer issue.
 1. Uncontrolled rolls in 99% is a hardware problem usually caused by motor/ESC issue, power failure or no signal ground on wires from FC to ESCs.

RBirdie001, 2017-03-05:
Thanks for so many interesting ideas!
My setups are:
1) Tricopter 32cm motor-center 9x4.7props,1200kV motors, 20A OPTO BlHeli ESC, FC NAZE32 rev. 6 10DOF.
This wasn't really serious issue, just at baro boosted throttle during tilt significantly, copter got much more side speed than I expected so it was difficult to stabilize copter without hitting trees around. Just possibility to limit maximal allowed influence of althold into throttle would help.
2) Quad 47cm M2M across, 9x4.7props,1200kV motors, 16A OPTO BlHeli ESC (connected only signal wire, for ground is used power ground 20cm to central ground point. I wanted avoid grounding loops).
FC Afro Naze32 Mini with external MAG. No blackbox logs, sorry...
Config dump is here:
[quad450_INAV16RC2_dump.txt](https://github.com/iNavFlight/inav/files/819287/quad450_INAV16RC2_dump.txt)
I was flying it for some weeks before without any problem, even tested full manual TH fast climbs, posholds, altholds  and several (at least 10 times) switch activated RTH - all worked perfectly. (Navigation modes made HUGE improvement in iNAV, big thanks!!!)
I only wanted make copter returning home faster so I increased set nav_max_speed = 500. After this change it was first RTH test, which ended immediately with crash.
I examined copter after crash (soft field - not much damaged copter) and didn't find any obvious hardware problem. All motors and ESC work, all sensors OK... After replacing prop and replacing one arm copter hovers again - real flight not tested yet.
I can't confirm it was really baro issue, but I thought it can have something common with problem 1) which I consider as a baro isuue.
Thanks for any ideas!
Roman

digitalentity, 2017-03-05:
For 1 - it's hard to tell without a blackbox log. NAZE32 Rev6 should have a dataflash so you can record some logs of the issue. Looks like an accelerometer resonance issue to me - need logs to confirm.

For 2 - this is easily testable - hover manually and give full throttle (manually). High possibility that your quad will flip. This is caused by noise picked up by long ESC signal wire at high throttle values. Signal wire coming to ESC is twisted with the ground for a reason - and that reason is noise tolerance.

RBirdie001, 2017-03-05:
Thanks!
I just wanted to let you know what I found to help the developement.
I'll keep testing and post if I find anything interesting. Still not think it was ESC wire interference - as I wrote, I have tested full TH climbs before without any problem. If nobody else reported such problem, it could be something with my hardware.
Roman

RBirdie001, 2017-03-07:
New observation:
In short: I found that neither baro nor navigation engine is the reason for roll problem described in 2).
In details:
After reparing the quad, checking all hardware and loading final iNAV 1.6 release I went fly again. After about 30 sec. of slow flying in horizon I tested full throttle fast uprise to about 30m. Copter was still behaving OK, but then at fast diving (I guess about 3m/s) I applied close to ground about 75% throttle to break descend. Copter was shaky as usual during vertical descend and then (still in manual-horizon) made the same diagonal barell rolls as previously in video 2) and crashed to ground. (Unfortunately on hard pavement so damage is worse than last time.  :-(  )
There wasn't althold nor navigation active at the time so it's definitely not the reason for problem 2) and I'm sorry for the confusion.
It really looks more like ESC or signals problem.
Tests about 1) I haven't done yet.

vic2k, 2017-03-10:
I had have the same issue when i switched from inav pos hold + inav alt hold to RTH. Kwad harshly turned front side to RTH point and start to wobbling and fast decrease altitude. I save kwad from crash by fast toggling from RTH to Angle.
F450
30A simonk
1000KV
NAZE32 rev5 10DOF

# tasks
Task list         rate/hz  max/us  avg/us maxload avgload     total/ms
 0 -       SYSTEM       9       9       5    0.5%    0.5%         9
 1 -          PID     499    1510    1350   75.8%   67.8%    133147
 5 -       SERIAL      99    7002       3   69.8%    0.5%        87
 6 -       BEEPER      99      12       4    0.6%    0.5%        81
 7 -      BATTERY      49   40104       9  197.0%    0.5%       130
 8 -           RX      49     105      94    1.0%    0.9%       941
 9 -          GPS      24     340      75    1.3%    0.6%       350
10 -      COMPASS       9     237     231    0.7%    0.7%       454
11 -         BARO      99     244     235    2.9%    2.8%      4628
Task check function            10       3                        28
Total (excluding SERIAL)                   279.8%   74.3%

# status
System Uptime: 179 seconds, Voltage: 122 * 0.1V (3S battery - OK), System load: 0.04
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n, BARO=MS5611, MAG=HMC5883
Sensor status: GYRO=OK, ACC=OK, MAG=OK, BARO=OK, SONAR=NONE, GPS=OK
Stack size: 2048, Stack address: 0x20005000
Cycle Time: 2004, I2C Errors: 0, config size: 1688

# dump

# version
# INAV/NAZE 1.5.1 Jan  5 2017 / 11:29:36 (556168d)
# pflags
# Persistent config flags: 0x00000000
# dump master

# mixer
mixer QUADX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -UNUSED_1
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -UNUSED_2
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
[... 306 lines cut ...]
set max_angle_inclination_pit = 300
set gyro_soft_lpf_hz = 60
set acc_soft_lpf_hz = 15
set dterm_lpf_hz = 40
set yaw_lpf_hz = 30
set yaw_p_limit = 300
set iterm_ignore_threshold = 200
set yaw_iterm_ignore_threshold = 50
set rate_accel_limit_roll_pitch = 0
set rate_accel_limit_yaw = 10000

# dump rates

# rateprofile
rateprofile 1

set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 45
set thr_expo = 15
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set tpa_rate = 0
set tpa_breakpoint = 1500

digitalentity, 2017-07-09:
@RBirdie001 anything new here?

RBirdie001, 2017-07-11:
> Komu: iNavFlight/inav <inav@noreply.github.com>
> much causing over controls. (#1359)
> "

Hi!
I tested big tilts in althodld mode on my tricopter and didn't see any 
problem. Quad, where I observed it is since that time unflyable so I can't 
verify it on it, but I think it will be OK even here and problems were 
caused with ESCs. According me this issue can be closed as fauolt in the 
code wasn't confirmed.
Thanks for the help!
Roman
---------- Původní e-mail ----------

(https://github.com/iNavFlight/inav/issues/1359#issuecomment-313919615), or 
mute the thread

"

DzikuVx, 2017-07-11:
@RBirdie001 Thanks for recheck, closing.

closed 2017-07-11 by DzikuVx