iNavFlight/inav #1360: NAZA GPS Error count steadily increases but data seem valid.

dbechth, 2017-03-05:
I recently damaged my original UBLOX gps and had a NAZA GPS clone laying around so I thought I would use that instead. I have built the firmware for my Naze32 Rev6 to include the NAZA protocol. I seem to be getting valid data but I have a steadily increasing error count as shown by the GPS tab in configurator.  Should I be worried? See screenshot and dump below.

When I built I added the following to target.h in the Naze folder...
`#undef GPS_PROTO_UBLOX
#undef TELEMETRY_LTM
#undef TELEMETRY_FRSKY
#undef USE_RX_PWM
#undef USE_SERIALRX_SBUS       // Very common protocol

#define GPS_PROTO_NAZA
#define TELEMETRY_IBUS`

[image]

`# dump
# INAV/NAZE 1.6.0 Mar  5 2017 / 02:52:10 (0def456)
mixer QUADX

mmix reset

servo 0 1000 2000 1500 90 90 100 -1
servo 1 1000 2000 1500 90 90 100 -1
servo 2 1000 2000 1500 90 90 100 -1
servo 3 1000 2000 1500 90 90 100 -1
servo 4 1000 2000 1500 90 90 100 -1
servo 5 1000 2000 1500 90 90 100 -1
servo 6 1000 2000 1500 90 90 100 -1
servo 7 1000 2000 1500 90 90 100 -1
smix reset

feature -RX_PPM
feature -VBAT
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature RX_PPM
feature VBAT
feature SOFTSERIAL
feature GPS
feature BLACKBOX
feature PWM_OUTPUT_ENABLE
beeper GYRO_CALIBRATED
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
[... 269 lines cut ...]
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 40
set pitch_rate = 40
set yaw_rate = 20
set tpa_rate = 0
set tpa_breakpoint = 1500

# `

digitalentity, 2017-03-06:
Don't have a NAZA GPS to verify. @sambas IIRC you own a DJI GPS module - can you please confirm?

sambas, 2017-03-07:
I can't reproduce this with my GPS. Issue sure is valid as friend of mine had this also. There might be some variation with message timings on cheaper and more expensive modules.
He had the cheaper clone:
http://www.banggood.com/Ublox-M8N-GPS-Compatible-with-DJI-NAZA-Lite-V1-V2-Flight-Controller-Phantom-1-2-Vision-p-991718.html
I have:
http://www.banggood.com/FFOSD-G2-M8N-GPS-Compass-Module-For-DJI-NAZA-M-V2-LITE-Flight-Control-p-979718.html

digitalentity, 2017-03-07:
For NAZA GPS we log an error if packet length is too big (>256 bytes) or checksum error. I'm almost certain this is related to GPS module itself, still need to figure out what it is sending

OlivierC-FR, 2017-03-07:
I have a genuine DJI Naza GPS, plugged on a SPRF3, and get zero errors.
http://i.imgur.com/juIcYms.jpg

digitalentity, 2017-03-07:
So, probably a hardware issue. If that GPS (with errors) is in fact working fine - actual failing hardware will be required to troubleshoot.

dbechth, 2017-03-07:
The one I have is a cheap clone from ebay that looks much like the first link mentioned by @sambas. I have been searching around and found [this ](https://forum.librepilot.org/index.php?topic=449.msg18189#msg18189). I'm thinking that this could be what is going on in my case. I will take a look at some raw data tonight and confirm.

digitalentity, 2017-03-07:
@dbechth great! Let us know what you find.

dbechth, 2017-03-10:
GPS stopped communicating though during testing but ublox chip still sending data. Closing this issue because I believe the issue to be with the cheap GPS and not iNav.

closed 2017-03-10 by dbechth