iNavFlight/inav #1361: Compilation fails if USE_SERIALRX_SPEKTRUM is not defined

giacomo892, 2017-03-05:
I've tried to compile a custom version for my NAZE target without Serial RX enabled since I don't use it.
In target/common.h I can compile if:

#define USE_RX_PWM
#define USE_RX_PPM
//#define SERIAL_RX
#define USE_SERIALRX_SPEKTRUM   // Cheap and fairly common protocol
//#define USE_SERIALRX_SBUS       // Very common protocol
//#define USE_SERIALRX_IBUS       // Cheap FlySky & Turnigy receive

Still saving lot of space, but cannot compile if I remove SPEKTRUM support like this:

#define USE_RX_PWM
#define USE_RX_PPM
//#define SERIAL_RX
//#define USE_SERIALRX_SPEKTRUM   // Cheap and fairly common protocol
//#define USE_SERIALRX_SBUS       // Very common protocol
//#define USE_SERIALRX_IBUS       // Cheap FlySky & Turnigy receive

The error is:

Linking NAZE
/tmp/cc7PTBML.ltrans0.ltrans.o: In function `init':
/home/giacomo/Scrivania/inav/./src/main/fc/fc_init.c:219: undefined reference to `spektrumBind'
collect2: error: ld returned 1 exit status
Makefile:846: set di istruzioni per l'obiettivo "obj/main/inav_NAZE.elf" non riuscito
make: *** [obj/main/inav_NAZE.elf] Errore 1

PS:
Removing PWM support (which I don't use gives a warning about an used variable but I think that's ok).

stronnag, 2017-03-05:
Maybe there are some SPEKTRUM bind defines that need to removed.

giacomo892, 2017-03-05:
USE_SERIALRX_SPEKTRUM  should disable completely SPEKTRUM so there is something nasty about this :)

DzikuVx, 2017-07-04:
@giacomo892 I was able to compile NAZE with following defines
```
#define USE_RX_PWM
#define USE_RX_PPM
// #define SERIAL_RX
// #define USE_SERIALRX_SPEKTRUM   // Cheap and fairly common protocol
// #define USE_SERIALRX_SBUS       // Very common protocol
// #define USE_SERIALRX_IBUS       // Cheap FlySky & Turnigy receivers
```

No errors, so I assume we can close it

closed 2017-07-04 by DzikuVx

giacomo892, 2017-07-04:
Yes. Now it works. Don't really know why.

@DzikuVx try to build with:

```
#define SERIAL_RX
#define USE_SERIALRX_SPEKTRUM   // Cheap and fairly common protocol
//#define USE_SERIALRX_SBUS       // Very common protocol
//#define USE_SERIALRX_IBUS       // Cheap FlySky & Turnigy receivers
```

same as

```
#define SERIAL_RX
//#define USE_SERIALRX_SPEKTRUM   // Cheap and fairly common protocol
//#define USE_SERIALRX_SBUS       // Very common protocol
#define USE_SERIALRX_IBUS       // Cheap FlySky & Turnigy receivers

```
you will get an error (but compiles) . maybe you can easily fix it :) 

thanks

DzikuVx, 2017-07-04:
Those are only warnings, not errors. I suppose we can live with them for a while

giacomo892, 2017-07-04:
ok, maybe we should rename this issue just to use it as todo :)