iNavFlight/inav #1362: Using a minim osd on a SP3 on UART3?

BoLtWD, 2017-03-05:
I am using a SP3 with inav and want to move the minim osd to a free UART uart3 port, 
This is so I dont have to disconnect each time I want to connect the USB port.

The inav info I found says you can have two MSP ports.
I have set uart3 port to MSP and have to correct connections to RX TX but no matter what I do I cant get it working. 

If I move the minim osd back to the TR, RX of UART1 all works ok.

Do you have any advice to where I am going wrong?﻿

digitalentity, 2017-03-06:
Please send a photo of your wiring and a screenshot of your Ports tab.

BoLtWD, 2017-03-06:
Hi,

Thank you for looking into this so confused to why this is not working.

I have done a wiring diagram as the wires on the bench are a birds nest. 

I moved the smart port off the X4R to a soft port so I could fee up UART3 
Then moved the OSD from UART1 to UART3 so I did not have to disconnect the OSD when connecting the USB.

I  know I can still get it working by connecting the OSD to UART1 but I am hoping to get this working so in the field I can connect the USB without removing wires. 

Again thank you for anything you advise on 

Wynne Davies (BoLt)

From: Konstantin Sharlaimov 
Sent: Monday, March 6, 2017 7:22 AM
To: iNavFlight/inav 
Cc: Wynne Davies ; Author 
Subject: Re: [iNavFlight/inav] Using a minim osd on a SP3 on UART3? (#1362)

Please send a photo of your wiring and a screenshot of your Ports tab.

BoLtWD, 2017-03-06:
Hi,
 
Thank you for looking into this so confused to why this is not working.
 
I have done a wiring diagram as the wires on the bench are a birds nest. 
 
I moved the smart port off the X4R to a soft port so I could fee up UART3 
Then moved the OSD from UART1 to UART3 so I did not have to disconnect the OSD when connecting the USB.
 
I  know I can still get it working by connecting the OSD to UART1 but I am hoping to get this working so in the field I can connect the USB without removing wires. 
 
Again thank you for anything you advise on 
 
Wynne Davies (BoLt)

[image]
[image]
[image]
[image]
[image]

BoLtWD, 2017-03-08:
I did what PAINLESS360 told me and wiped the FC to start again as sometimes this can fix this kind of problem. This was a good thing as it gave me a chance to update the firmware at the same time.
I enabled MSP on UART3 but this did nothing (same fault).

**Using below I have an interesting update on the problem.**

I swapped around the GPS that was on UART2 to UART3 and moved the OSD that was on UART3 to UART2. I set up UART2 for MSP and the OSD worked with the USB wire plugged in. 

Ok that’s the good bit. 

The GPS now on UART3 with the same setting it had on the other port did not work and I got a RED GPS informing me that there was a problem. So now this had stopped working, Grrrr

Just in case I changed TX and RX round and got the same. 

I swapped the GPS back round to UART2 with its settings and this again worked ok. 

So form all this I have no choice but to connect the OSD TO UART1 and remove the OSD when connecting the USB. 

I know UART3 works ok as originally I had my telemetry TX and RX connected to this at the same time UART2 had the GPS working and this all worked ok.  

This looks like its some kind of firmware clash port clash when using two UARTS ports as MSP and another UART for somthing like GPS or OSD

DzikuVx, 2017-06-09:
Most possible explanation is that SPRF3 UART3 Vcc is 3.3V from small onboard regulator. Probably GPS consumed too much current for that LDO to handle it.
I can confirm multiple MSP working together in latest INAV, so I'm closing this one

closed 2017-06-09 by DzikuVx