iNavFlight/inav #1363: Can't build iNAV 1.6 in Windows for target NAZE

RBirdie001, 2017-03-05:
Some months ago I successfully installed building enviroment in Windows 7 exactly as described here: https://github.com/iNavFlight/inav/blob/master/docs/development/Building%20in%20Windows.md
(I need to build myself due to non standard pinouts of my board, can't use ready .HEX)
All worked OK in iNAV 1.4, 1.5 and up to 1.6RC2, but from 1.6 build fails with the message:
...
test@test-PC /cygdrive/c/dev/inav1.6
$ make TARGET=NAZE
Linking NAZE
c:/dev/gcc-arm-none-eabi-4_8-2014q2/bin/../lib/gcc/arm-none-eabi/4.8.4/../../../../arm-none-eabi/bin/ld.exe: obj/main/inav_NAZE.elf section `.data' will not fit in region `FLASH'
c:/dev/gcc-arm-none-eabi-4_8-2014q2/bin/../lib/gcc/arm-none-eabi/4.8.4/../../../../arm-none-eabi/bin/ld.exe: region `FLASH' overflowed by 80 bytes
collect2.exe: error: ld returned 1 exit status
make: *** [Makefile:847: obj/main/inav_NAZE.elf] Error 1
...
Is there any help?
Thanks!
Roman

Schalonsus, 2017-03-05:
Try with newer gcc 6.2 or disable a feature, you are just creating a hex that is too large for naze target...

digitalentity, 2017-03-05:
Use newer compiler.

closed 2017-03-05 by digitalentity