iNavFlight/inav #1366: Take volts and amps values from MSP

sppnk, 2017-03-06:
As MWOSD can take variables from FC like RSSI and others, why dont allow the reverse thing? Could be done? (of  course FC should to be able to take them).

This is because in many setups the volts and amps are only in the OSD, but sometimes we fly the same airplanes w/o vtx, third view, and would be nice the FC to have this (for telemetry at least).

I have open the same request in mwosd repo.

What do you think?

tednv, 2017-03-18:
i have voltage + current in both osd and fc/gc/ez-gui in all my NAZE and F3 builds

sppnk, 2017-03-21:
Where are you measuring them? I do in my OSD, so I cant take the values to the FC. May be I should forget this and take always the reads in my FC's directly, but it is a more complex wiring.

tednv, 2017-03-21:
I use an apm power module with NAZE and F3 FCs (for current sensor and as UBEC). Wiring is simple, you just split the wire. One will go to the FC and one to OSD. For VBAT/VCC it's the same thing, you just have a wire going from battery to both and in OSD settings turn off FC voltage/current and use the readings directly in both FC and OSD. Current and Voltage inputs on OSD and FC are both analog voltage measurements, so you can send it to as many things as you want

sppnk, 2017-03-21:
Thanks for your advices, I have just finished the mod in my cc3d. Now I read the values directly in it.

Re.thinking on this, I guess it is better this way, so I dont know if the effort to modify MSP worths it.

So, I am fine now. Closing this.

closed 2017-03-21 by sppnk