iNavFlight/inav #1367: Trim ACC increase/decreases several decimals of degree in each stick touch (iNav 1.6)

raul-ortega, 2017-03-06:
I've seen something strange when tuning board alignment for ANGLE mode using Trim ACC sticks combinations. This is something very useful when we are at the field and we want to trim  ANGLE mode without the need of a computer neither a smartphone.

As shown in the following video that I've recorded, If I push the stick (roll/pitch) it some times increase/decrease 2 or 3, even more if it is hold it for a short period of time. 

I think that for a fine tuning of ANGLE once you are at field, without any screen showing the value each time you push the sticks, it would be a ideal behaviour if it increases/decreases only one decimal each  time the stick is pushed, otherwise you don't know how many decimals .

tednv, 2017-03-18:
Somewhat related:

[New Feature Request] display status - add board/accelerometer trim  #1425 

If the display were to show trim values you could tweak them up and down until you get the one you want, otherwise you're just blindly adjusting unless you've got a laptop with a wireless link, but at that point you might as well just use inav configurator to do it

digitalentity, 2017-03-19:
@raul-ortega it is designed to keep changing as long as you hold the stick, however it is updated according to receiver update rate. With some SerialRX (like IBUS which is updated as crazy rate) we can have it change a lot with one brief touch of the stick.

raul-ortega, 2017-03-19:
Mine is FrSky DR4-II, configured for PPM.
It is almost imposible to control how long shoyuld be hold the stick to get the minimun increment/decrement.

closed 2017-04-04 by digitalentity