iNavFlight/inav #1370: Error in documentation of fixed_wing_auto_arm parameter?

dropax, 2017-03-07:
Hope this is the right place to post this. ;) 

While getting familiar with inav I've found this nice feature. The description in 
https://github.com/iNavFlight/inav/blob/master/docs/Cli.md

> Auto-arm fixed wing aircraft on throttle above **min_throttle**, and disarming with stick commands are disabled, so power cycle is requirred to disarm. Requirres enabled motorstop and no arm switch configured.

states min_throttle as threshold, my poor understanding of the code implies that **min_check** is used which seems straight forward and common sense.

oleost, 2017-03-07:
You're right, it's is min_check.

If you know how you can do a PR to correct it. If not I will do it later.

dropax, 2017-03-07:
I'll probaly mess more up than fixing it, lol. 
Thanks for taking care of this. ;)

closed 2018-05-27 by stale[bot]

Dronek, 2019-05-29:
The error is still there.