iNavFlight/inav #1376: Error in docs/Board - Revolution.md

anbello, 2017-03-07:
From what I experienced on my quad and from the comparison between the file "src/main/target/REVO/target.h" and the Revolution schematic I saw the following errors:

- the section **RC_Input connector** (RX_PPM and RX_SERIAL) has to be split in two, one for the Revolution (10 pin input connector) and one for the Revolution Mini (8 pin input connector) and moreover it contains some inaccuracies, follows the correction:
**RC_Input connector (Revolution Mini)**
`RX_PPM and RX_SERIAL`
`Pin 	Function    Notes`
`1     Ground 	`
`2 	+5V 	`
`3 	PPM Input Enable feature RX_PPM`
`4`
`5     UART6 TX 	`
`6     UART6 RX 	`
`7`
`8`
`   `
**RC_Input connector (Revolution)**
`RX_PPM and RX_SERIAL`
`Pin  Function   Notes`
`1     Ground 	`
`2     +5V 	`
`3`
`4`
`5     PPM Input Enable feature RX_PPM`
`6`
`7     UART6 TX 	`
`8     UART6 RX 	`
`9`
`10`

- on the section **RC_Output connector** the last row has to be changed
`Pin  Pin    Function`
`6     PA0  MOTOR6 / RSSI_ADC`

- the connector **Pwr Sen** is undocumented
**PWR SEN connector**
`Pin Pin  Function`
`1          Ground`
`2          +5V`
`3    PC1 Current`
`4    PC2 Voltage`

digitalentity, 2017-03-13:
Can you share Revo and Revo Mini schematic?

anbello, 2017-03-13:
I only found the Revolution Schematic:
https://github.com/librepilot/LibrePilot/blob/next/hardware/Production/OpenPilot%20Revolution/Revolution%20Schematic.pdf

The Revo Mini is identical but don't has the radio module, the filters on input ports and has an 8 pin RC input port as described above.

closed 2017-04-05 by digitalentity