iNavFlight/inav #1377: Problem with RC input and OneShot ESC on REVO target

anbello, 2017-03-07:
On my Revolution Mini board with iNav 1.6 if I set normal PWM ESC I have no problems if I set OneShot ESC the RC input stops to work.
If I try to run the motors from the configurator is all OK so there is no problems with my ESC in OneShot mode.

digitalentity, 2017-03-13:
Are you using PWM or PPM RC input?

digitalentity, 2017-03-13:
Also, what target do you use? Does REVO target support Revolution Mini boards?

anbello, 2017-03-13:
I am using PPM RC input.
The Revolution Mini is identical to the Revolution, the only difference is that the Mini has no radio module, and so the target is REVO.
I tried both with the original 1.6 (internal compass) and with the little mod I asked you on the forum (external compass) and I have the same problem when I set OneShot.

For those who have not read about in the forum, the mod for external compass on external i2c bus is:
add:
#define MAG_I2C_INSTANCE I2C_DEVICE_EXT
to:
src/main/target/REVO/target.h

digitalentity, 2017-03-13:
@anbello thanks. This is now enough to try and reproduce the issue

digitalentity, 2017-04-05:
This is a hardware timer issue. On a REVO we have a lot of motor output - 11, and one of the outputs use the same timr (TIM12) that is used for PPM input.

anbello, 2017-04-05:
Is it possible to eliminate the use of the motor output that use TIM12?

digitalentity, 2017-04-09:
@anbello this is possible. I'll remove the clash.

anbello, 2017-04-09:
With this mods in file "main/target/REVO/target.h":
```
// Number of available PWM outputs
//#define MAX_PWM_OUTPUT_PORTS    11   // this is modified
#define MAX_PWM_OUTPUT_PORTS    6
#define TARGET_MOTOR_COUNT      6   // this is added
```
I have PPM RC input and OneShot ESC working together, but I think this is no news to you.

closed 2017-04-09 by digitalentity