iNavFlight/inav #1379: Feature Request: add support for CJMCU-530 i2c laser sensor to replace sonar

tednv, 2017-03-08:
For low altitude hover the i2c CJMCU-530 laser sensor would be much better than sonar for low altitude hover

Sonar has many issues especially transitioning between different surface types and changes in airflow

This sensor is inexpensive, tiny, low power consumption, is safe for your eyes, and there's already plenty of arduino code that simply has to be ported over to inav driver

There are some issues in cleanflight github mentioning expensive laser sensors, but no one wants to touch them in cleanflight code

code here:
https://codeload.github.com/pololu/vl53l0x-arduino/zip/master

I believe this could be a very valuable feature for inav

Linjieqiang, 2017-03-08:
This sensor looks great.

digitalentity, 2017-03-08:
I have one of those in the post office waiting to be picked up. @DzikuVx is testing his sample as so far is not very happy about it's performance.

digitalentity, 2017-03-08:
If it will function up to 1.5-2m *in daylight* - it will be a sonar killer

tednv, 2017-03-08:
I'm waiting for mine to come in also, will do some testing using arduino code first but from the specs it looks much better than sonar sensors. I ended up giving up on them and removing them from my builds since they are essentially useless when moving over changing surfaces or even solid concrete with slight changes in throttle. also since it's i2c may be able to use

this page has good description, should work well in bright daylight according to them
http://www.icstation.com/cjmcu-laser-range-sensor-module-rangefinder-distance-measurement-sensor-p-9907.html

digitalentity, 2017-03-08:
Sonars in general are not that bad - cheap ones like $1 HC-SR04s are crap, but more expensive ones are amazing. You need to mount them properly and process the data with sufficiently smart algorithms. 

Have a look at CONNEX Falcore drone - it uses sonar for altitude hold and it's perfect.

DzikuVx, 2017-08-03:
OK, since implementation is in place, I suppose we can close this one #1944

closed 2017-08-03 by DzikuVx