iNavFlight/inav #1386: Problem: arming

avirotstein, 2017-03-10:
Using iNav 1.6.0 on an Seriously Pro Racing F3 Delux board with a (properly reconfigured) "Walkera QR X350 Pro Quadcopter - GPS Module", will not arm, even after GPS fix is obtained, even after waiting many additional minutes.
Arm and disarm both result in a beep, so Rx setup is working. the GPS position shows correctly on my remote over telemetry, so that is configured correctly.
The 1 time it did arm (I don't know under what special condition this occurred), after lifting off it did not move forward after pressing forward on my remote so I quickly disarmed it, and could not get it to rearm. (my channel mapping is configured exactly as I did under Cleanflight)

digitalentity, 2017-03-10:
Please provide your configuration dump.

avirotstein, 2017-03-10:
Configuration is as follows:

2017-03-10 @ 11:26:08 -- Running - OS: Windows, Chrome: 56.0.2924.87, Configurator: 1.6.2
2017-03-10 @ 11:48:37 -- Serial port successfully opened with ID: 2
2017-03-10 @ 11:48:37 -- MultiWii API version received - 1.24.0
2017-03-10 @ 11:48:37 -- Flight controller info, identifier: INAV, version: 1.6.0
2017-03-10 @ 11:48:37 -- Running firmware released on: Feb 28 2017 13:44:32
2017-03-10 @ 11:48:39 -- Board: SRF3, version: 0
2017-03-10 @ 11:48:39 -- Unique device ID received - 0x3e002f5832570b20363436
2017-03-10 @ 11:48:40 -- CLI mode detected
Setup
Presets
Ports
Configuration
Failsafe
PID tuning
Advanced tuning
Receiver
Modes
Adjustments
Servos
GPS
Motors
OSD
LED Strip
Sensors
Tethered Logging
Blackbox
CLI
Note: Leaving CLI tab or pressing Disconnect will automatically send "exit" to the board. With the latest firmware this will make the controller restart and unsaved changes will be lost.

Entering CLI Mode, type 'exit' to return, or 'help'

# dump

# version
# INAV/SPRACINGF3 1.6.0 Feb 28 2017 / 13:44:32 (0def456)

# resources

# mixer
mixer QUADX

mmix reset

# servo
servo 0 1000 2000 1500 90 90 100 -1
servo 1 1000 2000 1500 90 90 100 -1
servo 2 1000 2000 1500 90 90 100 -1
servo 3 1000 2000 1500 90 90 100 -1
servo 4 1000 2000 1500 90 90 100 -1
servo 5 1000 2000 1500 90 90 100 -1
servo 6 1000 2000 1500 90 90 100 -1
servo 7 1000 2000 1500 90 90 100 -1

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
[... 431 lines cut ...]
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 80
set pitch_rate = 80
set yaw_rate = 65
set tpa_rate = 0
set tpa_breakpoint = 1500

#

digitalentity, 2017-03-10:
What is shown on main Configurator screen? Does all checks show green checkmarks?

avirotstein, 2017-03-10:
Everyuthing but navigation is safe (metal roof here, no portable computer at the moment to check what its like outdoors, though telemetry indicates a gps fix outdoors).
Strangely, if I set nav_extra_arming_safety=OFF then i can arm it indoors, but not outdoors.

digitalentity, 2017-03-10:
Red "navigation is safe" might indicate that position is not accurate enough to consider "safe" even if you have a valid fix. Also, you won't be able to arm if you already have PosHold, Waypoint or RTH enabled.

With `nav_extra_arming_safety=OFF` you should be able to arm regardless of GPS lock - it's something else that prevents arming in your case.

avirotstein, 2017-03-10:
I am arming via angle mode.

andre92, 2017-03-10:
Is your cpu overloaded maybe? Seems to be running 2k gyro, 1k pidloop and softserial. May I see the output from status command?

stronnag, 2017-03-10:
Very plausible. Perhaps two profiles, one for performance and no nav, the other lower performance and nav. 1k gyro, asynch=gyro and self levelling arms for me on SPRF3 with nav and soft serial.

avirotstein, 2017-03-10:
# status
System Uptime: 39 seconds
Voltage: 115 * 0.1V (3S battery - OK)
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n, BARO=MS5611, MAG=HMC5883
Sensor status: GYRO=OK, ACC=OK, MAG=OK, BARO=OK, SONAR=NONE, GPS=NONE
Stack size: 2048, Stack address: 0x10002000
I2C Errors: 0, config size: 3234, max available config: 4096
System load: 50, cycle time: 1048, PID rate: 954, RX rate: 47, System rate: 9

oleost, 2017-03-10:
Seems very high, share 'tasks'

digitalentity, 2017-03-10:
`System load: 50` is marginal. Just slight increase in system load and you won't be able to arm. System load won't grow until tasks actually start competing for CPU time meaning anything above just a few percent is potentially unsafe.

avirotstein, 2017-03-11:
I managed to get a wireless link working from my pc to outdoors. Confirmed 12 sats and navigation is safe is green.
With softserial off, cpu usage is 42-58%.
As for the gyro ratve. Changing from 2khz to 1khz gives 28% cpu usage with softserial on and 18% with softserial off. It now arms.
So it would seem the defauts are not usable on a "Seriously Pro Racing F3 Delux" board and perhaps changed?

digitalentity, 2017-03-11:
@avirotstein what "defaults" are you talking about?

avirotstein, 2017-03-11:
it seems to be the "5" racer" preset that sets the 1khz control look and the 2khz gyro sampling. well that profile is not seeming working on this particular board, with those values.

oleost, 2017-03-11:
If you read the description it's for f3 boards without any GPS. Only pure racing.

digitalentity, 2017-03-11:
@avirotstein that preset is for pure racing w/o GPS/baro. You are overloading the FC.

andre92, 2017-03-11:
your serial config: uart1 MSP, uart2 serial rx, uart 3 smartport, softserial gps.
Which baud is the gps using? Softserial should always run at the lowest baud possible, as more baudrate means more cpu load. Without softserial, your config would be no problem. 
I suppose you don't want to use ppm receiver mode to avoid the need for softserial. 
If you do wish to use softserial, try synchronous task processing at 1khz, that will improve navigation performance as well.

avirotstein, 2017-03-12:
Seems to be 5% cpu load when no presets are applied, and softserial enabled.

digitalentity, 2017-03-15:
Ok, I think we can close this.

closed 2017-03-15 by digitalentity