iNavFlight/inav #1389: Target: BETAFLIGHTF3

gregorg, 2017-03-10:
This is a great AIO board, can you add this target ?

digitalentity, 2017-03-11:
I need to have a look. I suspect there is no possibility to add compass there which will render it useless for quads with INAV; also it only has 4 outputs meaning it is a viable option only for flying wings with only 2 servos.

digitalentity, 2017-03-11:
Actually I2C can be made available if UART1 is removed. Then it will be possible to use it on a quad.

oleost, 2017-03-11:
Also remember you got DJI naza GPS for boards without i2C exposed.

gregorg, 2017-03-13:
Actually I want to use this board for a flying wing :)

digitalentity, 2017-04-29:
I'm postponing this to 1.8 in anticipation to make it along with  #1591

brkdwn, 2017-06-22:
I'am also interested for this target on a wing. I don't see it in 1.7. Is it still planned for 1.8 ? and then what is the time frame for 1.8 (not very aware of inav release  rythme)...

thx you for the great work !

airiclenz, 2017-07-20:
I would love to see the  BetaflightF3 as a target as well. Thanks for your awesome work!
/Airic

miginv, 2017-08-09:
I would also like to have it. Even without compass, I always like to use inav because GPS works great and iNav, but the same setup in Cleanflight or Betaflight doesn't work as good or even it is not detected. In my case I use GPS to have the coordinates in case I lose the drone far away. It already save my life once.

shellixyz, 2017-11-27:
I tried to add support for this board, feedback is welcome in PR #2514. You can find the code here https://github.com/shellixyz/inav/tree/BETAFLIGHTF3_target_complete

digitalentity, 2017-12-11:
Done

closed 2017-12-11 by digitalentity