iNavFlight/inav #1391: Disconnects and will not reconnect when saving from cli

CarefulNotToGetSuc, 2017-03-11:
Currently working on some custom mixes in CLI on a late 2011 15" MacBook pro, when I save and reboot in CLI and wait for the reboot to be confirmed complete, then switch to another tab, it acts as though no board is connected, and when I attempt to reconnect it refuses until I unplug the board and restart inav completely. CTO has tested this on his HP laptop running Windows 10 and has the same issue. Causes big delays in our work, as any time we wish to save changes in CLI we have to restart the entire system. Anyone else experienced this?

digitalentity, 2017-03-11:
I've noticed this as well. @DzikuVx maybe configurator-related (new msp queue code)?

CarefulNotToGetSuc, 2017-03-11:
We're running a v6 naze just for added info.

DzikuVx, 2017-03-11:
There is no queue in CLI tab. Or rather: there is no "drop" in CLI.
I had few cases like that on VCP boards. I'm using both Windows 10 and Mac for development. Either way, I will look into it.

oleost, 2017-03-11:
I have the same issue on two different Flip32 boards. My spracingf3 was fine.

DzikuVx, 2017-03-11:
@oleost rev 5 or 6? Which might be important. Configurator does some crazy
things during reconnect that are board dependent.

stronnag, 2017-03-11:
Flip 32, Arch Linux.

After `save` in the configurator CLI tab, it is necessary to restart the configurator to reconnect. 
Rebooting the board or resetting the serial device, e.g. `stty sane 9600 < /dev/ttyUSB0` do not help.

If the `save` command is issued from a serial terminal, the board reboots and the connection is retained.

oleost, 2017-03-11:
@DzikuVx Is there different revisions of Flip32 ?

DzikuVx, 2017-03-11:
Ok I suspect this is related with timings. Configurator tries to reconnect
after reboot before board is ready. I should have one Flip32 somewhere to
test.

DzikuVx, 2017-03-15:
Issue moved to Configurator project inav-configurator#170

DzikuVx, 2017-03-15:
Reopened since it is firmware issue after all. @digitalentity see comments in likned Configurator ticket

digitalentity, 2017-03-16:
@DzikuVx I'm on it.

DzikuVx, 2017-03-16:
fixed in #1417

closed 2017-03-16 by DzikuVx

sppnk, 2017-03-16:
Great, I had the same problem with cc3ds and nazes. Good job.