iNavFlight/inav #1392: 1.6.1 on NAZE F1 v6  gets unresponsive when enabling a quad preset

palkina, 2017-03-11:
After flashing the controller with 1.6.1 the system shows abnormally high CPU (588%) when selecting any quad preset.
I tried setting the preset for plane and it seems stable

palkina, 2017-03-11:
[nazef1v6_INAV1.6.1_CPUOverload.pdf](https://github.com/iNavFlight/inav/files/835880/nazef1v6_INAV1.6.1_CPUOverload.pdf)

oleost, 2017-03-11:
That preset is intended for F3 and newer CPUs. It will not work on an F1

closed 2017-03-11 by oleost

digitalentity, 2017-03-12:
@oleost @DzikuVx I wonder if we can report CPU family as an extra field in `MSP_BOARD_INFO` and check it when assigning presets.

DzikuVx, 2017-03-12:
It's somhow more comlicated since ale gyro bus comes in place. But yes, CPU family reposrting would be nice. At least we could say the user that this is probable to fail

oleost, 2017-03-12:
Also made an issue in regards to creating a preset that is suitable for 5" with GPS and across all CPU types.

https://github.com/iNavFlight/inav-configurator/issues/167