iNavFlight/inav #1393: [Feature Request]Add DShot ESC support.

Linjieqiang, 2017-03-12:
Port from betaflight.

oleost, 2017-03-14:
Any reason for closing this?

Dshot is more than just raw performance.
It gives to possibility in future to use motor as buzzers, etc etc.

Should be on the to do list for INAV.

Linjieqiang, 2017-03-15:
But I see DE is besy to deal with other problems so I choosed to close it.But I love to see dShot will be implemented in INAV project

fiam, 2017-09-25:
I'm moving this issue to 2.0, since there seems no interest in implementing DSHOT soon. PRs for it are welcome at any time though.

Azuxul, 2018-02-22:
Dshot implementation is important for BLheli32, without dshot we need to change throttle value and i dosen't found how to calibrate esc after. So for using inav with BLheli32 esc dshot is important.

Jeeper10A, 2018-03-17:
Is Dshot in the works? I have a Matek f7 board and decided on betaflight or Inav. INAV will be my first choice but only if Dshot is implemented,

Azuxul, 2018-03-17:
Currently Dshot doesn't work.

Blaster1920, 2018-05-09:
Any update on dshot1200 ?

Jhinta, 2018-05-11:
+1 waiting for dshot

strepto42, 2018-05-20:
+1 please, it's 2018 now. DSHOT is not only really useful, it's not even new tech now

digitalentity, 2018-05-23:
I see a lot of requests to add DHOT. Anybody want to have a crack at it?

turboworks, 2018-05-23:
You gave me all the servos, thank You very much, please+1 DSHOT me as well...https://youtu.be/koOqenMpvck

avrahams, 2018-07-11:
+1 and also esc telemetry

teckel12, 2018-07-28:
I wouldn't use DSHOT even it it was implemented, so I've got no motivation to add support. I feel there's too much potential risk with no gain over other more stable and tolerant protocols. But, there's a difference between a racing drone and a GPS navigation craft.

digitalentity, 2018-07-29:
DSHOT support is more a marketing move than a real added value. For racing quads it might be worth it (although I doubt the real effect on flight performance would be different from effect of a placebo).

For bigger machines it will definitely do more harm than good since it's much more succeptible for wiring quality and EMI noise.

strepto42, 2018-07-29:
I have to disagree. Dshot eliminates the need to calibrate escs and provides a much more fault tolerant protocol for control. Poor wiring and so on will actually affect non-dshot quads more. My long range rig suffers from motor twitch badly using one shot, but flies perfectly using Dshot (which is why I switched it back to betaflight). I have great respect for contributors to open source and this project in particular but seriously, get with the program guys. Digital > analogue

digitalentity, 2018-07-29:
On the wire DSHOT is analog. CRC checking masks lost/corrupt packets and what is observed as "motor twitching" with OS125 with DSHOT will result in spontaneous stop of one motor at some point.

teckel12, 2018-07-29:
@strepto42 Have you tried multishot? I've had issues with Oneshot early on with spontaneous full-throttle (of course to just one motor) which doesn't end well. Probably was an ESC issue, not INAV.

strepto42, 2018-07-30:
I agree that my issue is probably esc related. Still waiting for my spontaneous motor stop. Not gonna derail the thread further though. People want dshot, a tried and tested reliable protocol. The devs here can choose to implement it or not... There is no point arguing :-)

digitalentity, 2018-07-30:
DSHOT will be implemented eventually, but mostly because it'a a "cool thing".

turboworks, 2018-07-31:
ok, ok... so I was hoping for a fix ...https://youtu.be/ZF6K6MnojPg  ...https://youtu.be/SqGlyGRRors

teckel12, 2018-07-31:
@turboworks Tried Multishot?

turboworks, 2018-07-31:
been working with the older BL Heli with low kv motors... just hoping you guys that actually  know  what you are doing, will get these 32bit systems up and running,  for us experimenters

teckel12, 2018-07-31:
I'm holding out for 64 but ESCs, cause that means they're twice as good.

TwoToneEddy, 2018-08-05:
As I see this feature request is open I've volunteered to look into this once the required timer has been added by digitalentity.

teckel12, 2018-08-07:
@owengrey On the wire it's analog, but it's converted to a digital signal on the ends, which then does the CRC check.  This is a quite common "digital" communication technique.

digitalentity, 2018-08-07:
We live in an analog world. There is no such thing as "digital signal", only "digital data" and "digital transmission" which means that binary data entering the transmitter is identical to binary data leaving the receiver. Even two logic elements connected by a simple wire operate voltage levels which they interpret as binary "1" and "0".

Same for DSHOT - it's a modulation technique to transmit a digital packet over analog wire. One largest drawback of digital transmission - it's usually all or nothing - communication works flawlessly even at very high levels of distortion. Beyond that it doesn't work at all.

RCnator, 2018-08-12:
Pardon my lack of knowledge regarding IT etc but I hear about this timer and digitalentity wrote that he will "add a timer DMA framework after we release 2.0 (similar to what is there in Betaflight) - it will make adding dshot easier". What does that mean? Will we have dshot with 2.0.1 or will it be som kind of add on to 2.0?

teckel12, 2018-08-19:
@digitalentity Please reconsider. Imagine the users we'll have if INAV includes DSHOT. 😵

teckel12, 2018-08-21:
Braces for DSHOT, the next SBUS disarm bug of 2018. Only difference is this time there's no fixing it.

TwoToneEddy, 2018-08-21:
@digitalentity I'd love to look into this once the timer has been implemented. I'm literally moving house this weekend though so please don't wait for me if there's some urgency to get it in there for the next release. I know allot of pilots who will be thrilled at the addition of D-shot.

digitalentity, 2018-09-01:
@TwoToneEddy I've started on TIM/DMA framework in #3833 (work in progress). Adding DSHOT on top of that should be fairly simple.

digitalentity, 2018-09-01:
Also: I want DSHOT to be only available on targets where ALL motor outputs can support it. Otherwise firmware should automatically fallback to i.e. OS125. All DSHOT-related stuff should be guarded with define `USE_DSHOT`.

digitalentity, 2018-09-01:
@teckel12 yes, we'll definitely see increase in bugreports. We'll figure out DSHOT policy once we have some statistics.

davidngrc, 2018-09-03:
can't wait to test this, the next step will be esc telemetry, LOL

wx4cb, 2018-09-06:
Any news on dshot support?

teckel12, 2018-09-14:
@wx4cb TwoToneEddy said he was planning on adding this once the timing stuff was added, which it has been. I'd ask TTE of the status.

teckel12, 2018-09-15:
@TwoToneEddy You're doing it wrong. When I move, I get the Internet before anything. 😁

TwoToneEddy, 2018-09-15:
@teckel12 Don't get me wrong, I fuming I don't have the internet yet, how am I supposed to make a cup of coffee

turboworks, 2018-09-15:
dial-up?

digitalentity, 2018-09-15:
No internet is really annoying. When I moved I got Internet even before the bed, table and microwave 😆 

On topic: timer refactoring is in progress and I do it with DSHOT in mind - so adding DSHOT would be quite trivial. 

Have a look at LEDSTRIP driver built on top of new timer framework https://github.com/iNavFlight/inav/blob/5ace4e6e59c3600bbe45dbd8ea9bf2ef1e17477a/src/main/drivers/light_ws2811strip.c - that's the complete driver, all low level hardware ops are done in timer code. DSHOT is very similar to LEDSTRIP protocol so implementation wouldn't be too different.

digitalentity, 2018-09-15:
F7 implementation is clumsy and not tested yet. I kind of miss elegacy of ST's StdLib - HAL is true monstrosity. HAL_LL is much nicer, however amount of work to rewrite everything is huge (but eventually I might consider doing it anyway).

teckel12, 2018-09-16:
These days, the Internet comes before food, clothing and shelter, because what's the point without the internet?

digitalentity, 2018-09-22:
Initial cut of DSHOT implementation - #3879

digitalentity, 2018-09-22:
As I said - on new timer framework implementation is trivial

TwoToneEddy, 2018-09-23:
So you've done it already! That was quick! Sorry I couldn't pull my finger out in time, once I'm up and running I really hope I can contribute.

digitalentity, 2018-09-23:
@TwoToneEddy yep, new timer framework made it so trivial that I decided to go for it. Basically these ~140 lines of code are DSHOT protocol: https://github.com/iNavFlight/inav/blob/51dff7c844fd7690e8da687cbc2dc993540804ef/src/main/drivers/pwm_output.c#L196-L299

and mixer output conversion to DSHOT scale: https://github.com/iNavFlight/inav/blob/51dff7c844fd7690e8da687cbc2dc993540804ef/src/main/flight/mixer.c#L187-L218

One thing that will make code easier to understand is mixer change to [-1;1] scale which I might go for next.

DzikuVx, 2018-11-12:
I think we can close it as done in 2.1

closed 2018-11-12 by DzikuVx