iNavFlight/inav #1399: Add "Inverted flight stabilisation" mode

ABLomas, 2017-03-13:
(this is feature request, suitable only for airplanes)
Can we have "lock" on different angles, not only for level flight?
For example - inverted stabilisation (if angle is > 170° or < 190° - try to level to completely inverted flight, HORIZON mode), or "vertical stabilisation"? Acrobatics would be easy task and FPV users will enjoy this too (yes, i used this before on other AP's).
This would require few more parameters:
 - mode switch OR feature to turn it on (should be "OFF" for most users)
 - adjustment angles - CLI params (for example, "angle_inv_corr=8" - additional "pitch-up" angle when inverted, because most airplanes will require ele-down position for level flight inverted)
Inverted flight can be done on most airplanes.
Vertical hover/climb depends on motor, but should be safe to use on most planes.
Not sure about "knife edge" or "lock on 90° angle" - this would be issue on flying wing configuration, so, probably requires additional configuration.

Alternative implementation (simple and safe) - allow only roll "level" when inverted (pitch on gyro-only), this would be simple and require no trimming/config variables, but no autonomous/scripted flight possible; same for "knife edge". Still require roll/pitch stab on vertical hover...

For example, how it's done in MP:
https://github.com/MatrixPilot/MatrixPilot/blob/0131505fa3d5b74ea190176758072997e9439db1/MatrixPilot/behavior.c#L83-L151
And it's configuration here: https://github.com/MatrixPilot/MatrixPilot/blob/f021fcd6d7f01d540dd0df2212f6a2a6d6007c61/Config/options.h#L131-L139

Dronek, 2017-03-16:
I'm not sure if the term "inverted flight stabilization" applies to what you are asking for. Stabiliziation in acro mode works quite good, also in inverted flight. Moreover, it is questionable whether iNav should provide acrobatic-assistent features while focussing mainly on navigation related functions :-/

There might be some exotic scenarios where inverted leveling could be applied beneficially. E.g. for inverted landing when antennae, linkage or cams are mounted in an exposed manner underneath a flying wing.
No idea if it's worth the increased software complexity though.

ABLomas, 2017-03-16:
Inverted stab. in "ANGLE" or "HORIZON" modes.
Yes, you can fly inverted on acro, but if you turn on ANGLE mode - aircraft will flip over.

closed 2018-05-27 by stale[bot]