iNavFlight/inav #1415: Is it possible to arm by sending out arm signal and then hold last valid value?

liutairan, 2017-03-15:
I am trying to use MSP as the wireless control and telemetry method. 

I am able to arm my quadcopter by sending out raw rc values (aux1 for ARM) to FC continuously or get data from FC separately, but I wish to be able to get data (mainly GPS and other flight status) while flying. 

The serial port will get too busy if sending out raw rc values and requesting data at the same time. So I wonder is it possible to send out arm signal and then hold the last valid value so that I only need to send out ARM signal once, then the rest time I can focus on getting flight data.

liutairan, 2017-03-15:
Currently once I stop sending out arm signal, FC will get to failsafe mode quickly.

rodizio1, 2017-03-15:
I have never used it to do other stuff with it, just sending msp raw rc packets. On an F1 based Afroflight rev 5 running Cleanflight I used 100 packets per second which worked fine without overloading the FC. I don't remember exact figures, but I think it was at about 200 or 300 packets per second when the FC became overloaded.

So I'd figure, sending only let's say 5-10 msp raw rc packets per second (just enough to keep it from going into failsafe) shouldn't affect other stuff like telemetry too much.

closed 2017-04-03 by liutairan