iNavFlight/inav #1416: "Heading Lock" not working for airplanes?

Schalonsus, 2017-03-15:
Have two planes where i have "Heading Lock" active with "Angle" and "Alt Hold" but it wont hold the heading...

digitalentity, 2017-03-16:
I'm going to remove heading lock and change MAG mode to work on compass-less airplanes.

Schalonsus, 2017-03-16:
Good to hear. 
Can it be that enabling heading lock atm makes stabilization worse or does it have no effect at all?

sppnk, 2017-03-16:
@digitalentity , what about a 'cruise' mode with 'true GPS' heading? I guess we talked on this some time ago. Could be done? (for fpv it would be a nice feature).

Redshifft, 2017-03-16:
Cruise, Alt + Heading locks is a better name for aeroplanes, I agree

Dronek, 2017-03-16:
> I'm going to remove heading lock and change MAG mode to work on compass-less airplanes.

It would be nice to have multiple mode options like heading hold relative to north or track hold.
A more sophisticated feature could be to let the plane fly a straight track in the direction of heading at the time of activation. In other words: point the plane at e.g. a landmark, activate the feature/mode and it will undertake the necessary corrections to possibly fly there on a straight path.

[image]

raul-ortega, 2017-03-19:
What Dronek suggests could be done estimating a waypoint (whe know speed and heading) and fly towards it, and re estimating a new waypoint once it has been reached.

Dronek, 2017-03-20:
> What Dronek suggests could be done estimating a waypoint (whe know speed and heading) and fly towards it, and re estimating a new waypoint once it has been reached. Or perhaps estimating a waypoint as far as possible to be sure that the plane wont reach it.

I'm not familiar with the details how taveling from one WP to another is handled in iNav but in the worst case "just" trying to reach a point won't result in a straight path when crosswind is involved. Refering to the picture above, a straight track could be realized by minimizing crosstrack error / lateral deviation from the original course. 

Also, please note that setting a distant waypoint based on FPV sight (e.g. by using OSD center crosshair) would heavily rely on aircraft orientation (compass heading), rather than GPS heading.

digitalentity, 2017-03-20:
Please keep in mind that there is no distinct "compass heading" or "GPS heading" in INAV. There is just one "heading" which is assumed to be where aircraft nose is pointing.

Dronek, 2017-03-20:
> Please keep in mind that there is no distinct "compass heading" or "GPS heading" in INAV. There is just one "heading" which is assumed to be where aircraft nose is pointing.

Does that have implications on the choice of heading used in mwosd? I noticed that in strong headwind situation my OSD heading stayed steady while Olivier C. at RCG reported flipping of the OSD heading in such case. I deliberately chose to use MAG heading in mwosd config.h and am now wondering whether iNav is using "one heading" for autopilot or in general "one heading" that is also sent to the OSD regardless of OSD settings.

stronnag, 2017-03-29:
Aframe heading is heading, typically from the compass. If you want course over the ground, then you must calculate it (as mwp does for 'bearing', in mwp terms the compass direction between the vehicle and home)

sppnk, 2017-03-29:
Hmmm... I will steal your code for that. :-)

sppnk, 2017-03-29:
@Dronek , see here: https://github.com/ShikOfTheRa/scarab-osd/issues/404

Dronek, 2017-03-30:
@sppnk : interesting discussion, also at scarab-osd repo. I'm actually using the mentioned `define USEMAGHEADING` option in mwosd. 
@digitalentity : can we be sure that iNav will pass MAG heading to mwosd when using the mentioned option?

digitalentity, 2017-03-30:
@Dronek there is no "mag heading" or "gps heading" in INAV. There is only "estimated heading" which is essentially "mag heading" if compass is available, "gps heading" if compass is not available but there is GPS or "random heading" if neither is/was available.

Dronek, 2017-03-30:
>  There is only "estimated heading" which is essentially "mag heading" if compass is available, "gps heading" if compass is not available 

OK, so apparently it makes no difference if `define USEMAGHEADING` option is set in mwosd or not but at least I can be relatively certain that my OSD compass is mainly based on compass sensor, when I understand you correctly.

However, it would be of more value when compass data (MAG heading) and course over ground (GPS heading) were available for OSD to indicate crosswind, as I commented [here](https://github.com/ShikOfTheRa/scarab-osd/issues/404#issuecomment-290344139).

sppnk, 2017-03-30:
@Dronek , as I use cc3d, I dont have mag, so the value sent in LTM is not what I need to compute a reliable "direction to home", becasue I guess that value is an 'attitude heading'.

@digitalentity , I guess that 'GPS heading' (compass not available) is not equal to course over ground, but  plane attitude heading. 

If I move the plane sttoped in the ground that value moves, but the plane is at the same point. 

I guess that value comes from gyros, not GPS (please correct me if I am wrong).  

So I am testing this week the calculation of that 'course over ground' value in my own LTM oled arduino reader. I hope this will work. I will report on results.

digitalentity, 2017-03-30:
Gyros are always used to estimate heading. In absense of compass sensor GPS course is used as a reference = gyro estimated attitude heading is aligned with GPS course over ground. If GPS is temporary lost we still have valid heading because gyros will cover the gap.

sppnk, 2017-03-30:
@digitalentity ,then, with the plane flying, 'Heading' value from LTM frame 'A' should be aproximately  equal to 'course over ground'? (I cannot find easily that in the sources of inav).

digitalentity, 2017-03-30:
@sppnk if you don't have compass sensor - yes. If you have compass - it will show true heading (azimuth)

sppnk, 2017-03-30:
Thks @digitalentity , I will continue my tests this week capturing LTM telemetry when plane is flying and checking it in the workbench.

sppnk, 2017-03-31:
I have captured some LTM telemetry data, and I have computed 'angle to home' and 'course over groud', taking the gps values of the plane in flight with my small telemetry project.

When the plane is 400m far, I connect RTH and it comes almost straight to home.

Then, 'angle to home' and 'course over ground' are very similar, as they should be.

BUT the 'heading' value from LTM frame 'A' is not consistent with this. I guess it is the attitude, not the 'course over ground'. Could you check that in the code? I dont need it, as I am computing the value in my LTM reader, but I would like to be sure.

sppnk, 2017-04-02:
@digitalentity , @Dronek , I have captured again data after some bug corrections in my LTM reader code. And you are right: 'heading' value from LTM id perfectly the 'course over ground' when plane comes direct to home in RTH. 

So please forget this and sorry for the mess. :-)

digitalentity, 2017-04-03:
No worries!

Schalonsus, 2017-05-03:
Any news regarding heading hold for flying wings?
Without this, iNav is not really relaxing for longer flights....

sppnk, 2017-05-03:
@Schalonsus , I guess some work was done recently: https://github.com/iNavFlight/inav/pull/1433

digitalentity, 2017-05-03:
@Schalonsus, @sppnk heading lock is only for airplanes with rudder. Flying wings still require quite a bit of work.

closed 2017-05-24 by digitalentity