iNavFlight/inav #1422: New MSP commands for missions

eziosoft, 2017-03-17:
Could we have this implemented through MSP:
- possibility to get maximum supported number of waypoints
- possibly to get number of current waypoints in the memory
- possibility to trigger modes like RTH, PH, land...etc
- change behavior of "flags" in WP. (Currently flags are used only to detect that this is the last waypoint, and it has to be set by GCS. Maybe we can use it for some other stuff. What I would love to see is ie 4 bits used by inav for something, and four bits that are free to use by GCS.) Also having a possibly to get number of waypoints in memory signalization of the last WP through flags is not needed.

What do you think?

stronnag, 2017-03-17:
Having an MSP to get the waypoint info would be useful. I rather like having the sentinel flag to terminal the set of waypoints (and it works both ways for upload and download), but there's no need for it to be anything other than a single bit (however attractive alternating bit patterns were to esobandi). In future just using bit 7 for the sentinel would be OK.

digitalentity, 2017-03-30:
@eziosoft we can create additional MSP commands for 1) and 2)

eziosoft, 2017-03-30:
Cool. Thanks

astragalizont, 2017-11-23:
Hi, would it be possible to add camera trigger by WP/mission?

closed 2018-05-27 by stale[bot]