iNavFlight/inav #1423: [Feature support]Add airspeed support for plane.

Linjieqiang, 2017-03-17:


Dronek, 2017-03-17:
It is coming...
See [here](https://github.com/iNavFlight/inav/pull/386) and [here](https://github.com/iNavFlight/inav/pull/793).

Linjieqiang, 2017-03-25:
I will take some PRs in this https://github.com/iNavFlight/inav/pull/1456.

digitalentity, 2017-03-25:
@Linjieqiang I took the liberty to edit your issue description and change it to task list.

Linjieqiang, 2017-03-25:
@digitalentity Great.Come on.

ServError, 2017-04-04:
A number of us already own the EagleTree Airspeed V3 sensor from other projects, which reads two bytes over i2c in "raw" mode, providing a m/s reading using a formula which has been semi-verified here - https://www.rcgroups.com/forums/showpost.php?p=25357499&postcount=142

It also supports a "3rd Party" mode (http://www.eagletreesystems.com/Manuals/microsensor-i2c.pdf) in which i2c reads out in mph or kph, using the built-in self-correcting calibration, depending on what the user has set. There is no way to determine if it is in mph or kph the data though, so this gets complicated unless you poll the user for which mode it is in from the configurator.

Official documentation claims that the i2c address is 0xEA while some users report needing to use 0x75

From the pdf above, it seems a driver would be trivially easy to implement. Does anyone with a unit want to take a stab at it? If not, I'll try to get around to it at some point.

digitalentity, 2017-04-04:
@ServError now this is interesting. Yes, I believe it should be fairly trivial to implement - I would have a crack at it, but unfortunately I don't own a unit to debug with.

closed 2018-05-27 by stale[bot]