iNavFlight/inav #1425: [New Feature Request] display status - add board/accelerometer trim

tednv, 2017-03-18:
Add Accelerometer/Board trim values to the disarmed status screen (next to HDOP: value perhaps? or new line if there is room)

This would be a HUGE help in getting the trims set right out in the field without having to drag a laptop along, since you can just trim using the radio, take off, land, disarm, look at display on the drone and then trim again

Display shows stuff like Altitude and Heading, which aren't as useful as trim values that you can change using sticks

tednv, 2017-03-18:
I have this so far in my fork, it's working well but needs to be formatted and organized better:

commit 356588619eec458c5881719e9c29966b88a192f2

right now trim values (roll + pitch) do get updated realtime as you're adjusting them with the sticks but value needs to be divided by 10 and printed (signed)

also I added this as new line, when screen is busier with gps and mag data the last line may get cut off or scroll over the top

tednv, 2017-03-18:
If there is no room I think it would be worth replacing mag and baro values with trim instead, much more useful

digitalentity, 2017-03-18:
@tednv please submit a PR - this looks useful.

tednv, 2017-03-19:
I have a different PR open (display updates  #1424), can't seem to find a way to open a separate PR for this, I'm somewhat of a github noob will look for a way to make this work. I did test the code mention in the commit above and it works well, I just need some help getting tfp_sprintf to properly handle floats, will probably need a custom method to do this. also need some help fitting this into existing display components, there's not much room to work with

tednv, 2017-03-21:
this is all done and working in #1438

digitalentity, 2017-03-25:
@tednv thanks. I'll close the issue now, review and merge the #1438 soon.

closed 2017-03-25 by digitalentity

tednv, 2017-03-25:
added cleaned up PR based on development branch:

add accelerometer trim to display  #1457