iNavFlight/inav #1428: Auto land regression in 1.6.2 (681b611)

stronnag, 2017-03-18:
In 1.6.1 and earlier, auto land has worked well, descending rapidliy at first, slowing down and disarming on landing:
```
set nav_land_slowdown_minalt = 400
set nav_land_slowdown_maxalt = 1000
````
On 1.6.2, yesterday I noticed that landing was extremely slow (http://seyrsnys.myzen.co.uk/inav-162-landing/LOG00001.TXT) (600mm tricopter). The descent from 24m to poweroff takes c. 70 seconds.

Today I tried it on a 420mm quad. Again, same symptoms, extremely slow initial descent. The machine then auto-disarmed at 16.4m; resulting in a more rapid descent, though not the intended effect. Fortunately the ground is extremely soft and the machine fell level, resulting in no damage. Unfortunately, the logger was not working.

The machine was reflashed to 1.6.1 and flown again. Auto land works perfectly. http://seyrsnys.myzen.co.uk/inav-162-landing/LOG00633_2.TXT ; the descent from 21m to power off is c. 12s.

I'm sorry for not having a log of the disarm at 16m incident; I must have failed to reconnect it after flashing 1.6,2. I'm also reluctant to try and get a log, as I doubt I will be so lucky a second time around.

stronnag, 2017-03-18:
As I've been noticing annoying (but not fatal) slowdown in autoland, I decided to see how 1.6.x has performed. Other than the worst performing (in very slow) is 681b611, there's not much correlation and a large variation. I probably have BB logs for most of these (other than the above interesting one). Attached is a spreadsheet (CSV) ordered by autolanding rates in m/s for all 1.6.x tested (some of the git ids may be private merges which could be resolved).

[landing.txt](https://github.com/iNavFlight/inav/files/852696/landing.txt)

digitalentity, 2017-03-18:
@stronnag is that machine equipped with sonar? IIRC your sonar is reading 30-40cm all the time and autoland code will switch to slow landing if sonar is reading < 50cm.

digitalentity, 2017-03-18:
Disarm in the air is more worrying though - probably caused by too low threschold for throttle.

stronnag, 2017-03-18:
The tricopter which also suffers from slow land in 681b611 is not equipped with sonar. 

The quad has a different sonar sensor from last time (which I wanted to log!), but sonar was not enabled at the time of disarm.

So it looks possible that the sonar related changes are causing slow land unconditionally???

It was also quite windy, so the machine was being blown around a bit, 0.5s disarm timer is probably not too wise :)

digitalentity, 2017-03-18:
Okey, I'm now looking at your tri logs. Might be sonar related changes
indeed...

stronnag, 2017-03-18:
Ho ho, this will not be the first time I'm had a disarming result from sonar experiments. Still remarkable no damage falling 16m.

digitalentity, 2017-03-18:
Found it. It was indeed new sonar code - wonar was always reported valid if enabled at compile-time. Apparently that was my doing :smiley:

digitalentity, 2017-03-18:
@stronnag fix ready https://github.com/iNavFlight/inav/pull/1430

stronnag, 2017-03-19:
Thanks. I will test it again as soon as possible, may not be today.

digitalentity, 2017-03-19:
Take you time. Thanks

stronnag, 2017-03-19:
Confirmed fixed on two different machined, 600mm tri and 280mm dead cat. Thanks.

digitalentity, 2017-03-19:
Cool! Fix merged!

closed 2017-03-19 by digitalentity