iNavFlight/inav #1435: AutoLauch - fixed wing - not reaching climb angle

Schalonsus, 2017-03-19:
Very rare times it is not working. 
The AHI on the OSD schows the same distance from the middle on the screen on failed and successed launches, so there must be some board alignment initializing problems or so. Because if you look at real horizon it is way different on failed and successed launch. 
https://youtu.be/JYsVpsRqQI4

digitalentity, 2017-03-19:
Hard to debug without BB logs.

oleost, 2017-03-19:
Just a wild guess, try and lower motor throttle if there is example some vibration that messes with accelerometer.

digitalentity, 2017-03-25:
That would be my guess as well - some sort of resonance that is messing up the acceleration and cause attitude drift.

Schalonsus, 2017-03-25:
I changed to a folding prop, that should make even less vibration... We will see

oleost, 2017-03-25:
If I'm not mistaken you have carbonfiber frame on that wing? You should probably think of softmounting with some gyro tape or similar

Schalonsus, 2017-03-25:
Yes its carbonfiber.
Here are some logs, dont know if gyro/acc data is clean, you can look yourself:
[12deg_launch1.TXT](https://github.com/iNavFlight/inav/files/870059/12deg_launch1.TXT)
[12deg_launch2.TXT](https://github.com/iNavFlight/inav/files/870060/12deg_launch2.TXT)
[24deg_launch2.TXT](https://github.com/iNavFlight/inav/files/870061/24deg_launch2.TXT)

Landflieger, 2018-05-13:
Any news with this issue? have sometimes the same problem on my bix3

digitalentity, 2018-05-13:
Problem is linked to possible horizon drift in launch, which in turn was caused by bug in OSD driver (fixed now). If you still have the issue, please provide logs confirming.

closed 2018-07-27 by stale[bot]