iNavFlight/inav #1441: RCExplorerF3 Baro not detected

nwenkel, 2017-03-20:
onboard Baro on RCExplorerF3 board (RCExplorer MiniTri F3FC ) is not detected when magnetometer is connected  to I2C. Any comments would be perfect.

Ralf-W, 2017-03-20:
This might help: https://github.com/iNavFlight/inav/issues/438 

> wilco1967 commented on 17 Aug 2016 • edited
> Ok... found out the problem was the pull-up resistors on the external mag board,
> 
> Replaced the compass with another board I had lying around, which doesn't have pull-up resistors, and now both compass and baro work just fine together.
> 
> They also work together in Triflight.
> 
> So the problem was too many pull-up resistors....
> there is 10kOhm on the F3FC, and the compass also had 10kOhm.... that was apparently too much combined....

nwenkel, 2017-03-22:
@Ralf-W 
Thanks, I´m using the BN-880 GPS/Compass modul and according to some web information it should not have the pull up resistors. Any comments how to solve this issue, e.g. to remove the pull up on the F3FC board would be helpful. By the way using iNav 1.6 the detecting works but after a while i got a lot of I2C errors.

digitalentity, 2017-07-09:
I'm closing this issue since there's no activity for quite a while. Feel free to reopen if necessary.

closed 2017-07-09 by digitalentity