iNavFlight/inav #1442: RMDO Dodo

Ernie34, 2017-03-20:
I see the RMRC Dodo is on the end of life list? Why? It's not hard to get from RMRC, It's got all the same features and isn't slated to be discontinued anytime soon. 

http://tinyurl.com/jqtxo7z

oleost, 2017-03-20:
Maybe because rev 1, and 2 had small flash size.

stronnag, 2017-03-20:
The table seems (and now admits) to being more about sponsorship that actual testing. My EVO gets tested more than most of the other boards combined. "not well tested" my arse.

oleost, 2017-03-20:
@stronnag The sponsored boards are clearly labeled.

The wiki is open for _everyone_. If you think EVO is great board feel free to add it to the recommended ones.

I added spracingf3 because that's what I use and it support wide array of features including channel forwarding and servo gimbal.

I also wanted to add the spracingf3mini, but unsure how well tested it is. And how it's works with sonar and servo gimbal.

digitalentity, 2017-03-20:
@stronnag I put the EVO to not well tested boards because reported failure rate for EVOs is quite high. The boards that survive - work perfectly good.

Ernie34, 2017-03-23:
The Dodo has not changed at all. It does seem that this is all sponsorship
based which kills the credibility of it. A's long as you're buying a REAL
dodo and not a knockoff it should be perfect for this firmware.

oleost, 2017-03-23:
https://github.com/cleanflight/cleanflight/blob/master/docs/Board%20-%20RMDO.md

> The revision 3 board changed the CPU from one with 128KB to one with 256KB however, for compatibility reasons, Cleanflight only supports and uses 128KB of it. The same binary can be used on all DODO boards

oleost, 2017-03-23:
Also the list does not hide which board are sponsored, also since it's on the wiki everyone is welcome to add boards to the recommended ones if they see it fit. As already pointed out.

martinbudden, 2017-03-23:
It does annoy me a bit when people start saying "it's about the sponsorship". It's more about the practicalities of things.

There are over 50 different flight controller boards available and iNav cannot support them all to the same degree. I spend quite of a bit of my own time developing iNav software and only own a handful of different board variants (well 10 actually) - I test on these. Of those 4 types were donated by what you might call "sponsorship", the rest were purchased by myself.

I would have expected the board manufacturers themselves to be a bit more pro-active: testing their boards on iNav and either confirming they work on the latest release or submitting bug reports, but by and large that does not happen.

If you'd like to see the RMRC board better supported by iNav then there are at least 3 things you could do:

1. "Adopt" the board yourself. That is become the iNav "go to" person for the board. Test the board when new versions of iNav come out, submit bug reports if you find bugs, and update the wiki pages with information about the board.
2. Complain to the manufacturer and ask them to actively support iNav on the board, by doing the things suggested in (1) above.
3. Donate a board to someone who is willing to do the things in (1) above.

And @stronnag , you do so much testing that I fully support you changing the wiki to add any board you actively use as a recommended board.

digitalentity, 2017-03-25:
Although it's close to impossible to get original RMDO and good quality EVO boards these days, I added EVO and RMDO to recommended boards as they are indeed very well tested (especially by @stronnag).

Ernie34, 2017-03-26:
Not hard at all to get a RMDO board at all. Just order it from ReadyMadeRC.

digitalentity, 2017-07-09:
I believe we can close this issue. RMDO boards are supported by SPRacingF3 firmware.

closed 2017-07-09 by digitalentity