iNavFlight/inav #1443: Built in OSD (omnibus f3): "Direction to Home" arrow doesn't point to home.

raul-ortega, 2017-03-20:
The flight controller is aligned 270º in yaw axis. No compass is present.
In this video before take off and during approximation to home before landing you can realize that the arrow doesn´t point to home.

(https://www.youtube.com/watch?v=hqOCHuA-ISE)

raul-ortega, 2017-03-20:
[simba-1.6.1-19.03.2017-diff.txt](https://github.com/iNavFlight/inav/files/856511/simba-1.6.1-19.03.2017-diff.txt)
[simba-1.6.1-19.03.2017-dump.txt](https://github.com/iNavFlight/inav/files/856512/simba-1.6.1-19.03.2017-dump.txt)

digitalentity, 2017-03-20:
Try uploading this font
[default.mcm.gz](https://github.com/iNavFlight/inav/files/856625/default.mcm.gz)

raul-ortega, 2017-03-21:
Not tested yet but iNav source code says this:

// Direction arrows
#define SYM_ARROW_SOUTH 0x60
#define SYM_ARROW_2     0x61
#define SYM_ARROW_3     0x62
#define SYM_ARROW_4     0x63
#define SYM_ARROW_EAST  0x64
#define SYM_ARROW_6     0x65
#define SYM_ARROW_7     0x66
#define SYM_ARROW_8     0x67
#define SYM_ARROW_NORTH 0x68
#define SYM_ARROW_10    0x69
#define SYM_ARROW_11    0x6A
#define SYM_ARROW_12    0x6B
#define SYM_ARROW_WEST  0x6C
#define SYM_ARROW_14    0x6D
#define SYM_ARROW_15    0x6E
#define SYM_ARROW_16 0x6F

And the font says this:

[image]

marbalon, 2017-03-21:
Yes, but definition in header comes from old fonts, new one have different direction mapping and should be redefined in header file.

raul-ortega, 2017-04-01:
What does mean the label Cleanup? Does this mean that you have to rewrite the code to correct the bug?
I was planning to test it tomorrow, but if you consider that it is not necesary I won't upload the font.

digitalentity, 2017-04-01:
Cleanup means that character definitions in the code need changing. Arrow itself works fine with proper font.

raul-ortega, 2017-04-04:
I suspect that the problem is within the folowing osd.c code: 

..
case OSD_HOME_DIR:
        {
            int16_t h = GPS_directionToHome - DECIDEGREES_TO_DEGREES(**attitude.values.yaw**);
...

Does attitude.values.yaw take into account the board alignment in yaw axis? Mine is rotated 270 degrees.

digitalentity, 2017-04-05:
Wrong arrow direction is fixed by using proper font. Inconsistency between the font and code fixed by https://github.com/iNavFlight/inav/pull/1519

closed 2017-04-05 by digitalentity