iNavFlight/inav #1444: Built in OSD (omnibus f3): Battery level icon is always almost full

raul-ortega, 2017-03-20:
The icon representing the battery level always keep almost full.

[https://www.youtube.com/watch?v=hqOCHuA-ISE](https://www.youtube.com/watch?v=hqOCHuA-ISE)

raul-ortega, 2017-03-20:
[simba-1.6.1-19.03.2017-diff.txt](https://github.com/iNavFlight/inav/files/856509/simba-1.6.1-19.03.2017-diff.txt)
[simba-1.6.1-19.03.2017-dump.txt](https://github.com/iNavFlight/inav/files/856510/simba-1.6.1-19.03.2017-dump.txt)

digitalentity, 2017-03-20:
Try uploading this font
[default.mcm.gz](https://github.com/iNavFlight/inav/files/856625/default.mcm.gz)

raul-ortega, 2017-04-12:
The font by itself doesn't solve this issue. The pull request I have submitted solves it together with the font.
By the other hand, the closed issue [#1443](https://github.com/iNavFlight/inav/issues/1443) is solved by using the font you uploaded.

raul-ortega, 2017-05-30:
I re-open this issue because I have detected that the icon is full once the battery level is lower than the minimum voltage (lower than 9.9 volts for 3 cell battery, for example).

I suspect (not sure) that the problem comes from function calculateBatteryPercentage, which would be returning 100 for vbat levels lower than the minimum.

raul-ortega, 2017-05-30:
[https://youtu.be/8ufZkhPB6o4?t=747](https://youtu.be/8ufZkhPB6o4?t=747)

digitalentity, 2017-05-30:
Firmware has no option to configure battery cell size - cell count is calculated automatically by dividing measured voltage by `vbat_min_cell_voltage` (default 3.3V). Therefore voltages lower than 9.9 are miscalculated to be a fully charged 2S battery.

Solution - set `vbat_min_cell_voltage` and `vbat_max_cell_voltage` according to your battery chemistry and min/max voltages.

DzikuVx, 2017-08-06:
Tested INAV 1.7.2 and it works fine. Closing

closed 2017-08-06 by DzikuVx