iNavFlight/inav #1445: Built in OSD (omnibus f3): Artificial horizon some times doesn't match real horizon

raul-ortega, 2017-03-20:
Please, see attached picture:

[image]

stronnag, 2017-03-20:
You've seen the comment in https://github.com/iNavFlight/inav/blob/master/docs/Board%20-%20OMNIBUS.md about support for the OSD?

Or is this a different board, or the documentation out of date?

raul-ortega, 2017-03-20:
@stronnag do you mean the cooment "Integrated OSD is not supported yet"?
I think the documentation is out of date, if you see the picture I uploaded, the built in OSD is already supported for the omnibus target. Could be that it is not fully supported, but of course that it is already supported.

stronnag, 2017-03-20:
Yes, that comment. Just wondering .... digitalentity will be up soon, and he'll know for sure.
Bonus if it does work; I'll have to buy some FPV gear.

digitalentity, 2017-03-20:
I npoticed that as well. On boot it's fine, after launch if sometimes off, after a minute of flying it's fine again.  This is not OSD fault, the actual estimated attitude is calculated wrong.

Interestingly, this only happens on Omnibus F3 board and only on airplanes - I was unable to reproduce on other FCs. My guess it's faulty hardware design - power supply for gyro sensor is not clean enough and causes incorrect readings in certain conditions.

ServError, 2017-03-21:
Seems odd that a hardware design issue would only be apparent on planes? Makes me think of servo noise since that's the main difference between the two configurations. Are we talking about setups with servos running off of the on-board regulator or powered independently? If it is hardware related, I wonder if it's a matter of board design or easily dealt with by removing/bypassing the regulator.

raul-ortega, 2017-03-21:
Servos are powered from a external filtered bec, but is co nnected to the
same batery than FC.

Feldsalat, 2017-05-23:
Same problem here: 
[http://www.youtube.com/watch?v=ej4l6ltAtJU](url)
[http://www.youtube.com/watch?v=uYUYqOanh1k](url)
Inav 1.6.1 Omnibus F3 AIO

digitalentity, 2017-05-24:
From recent reports it appears to be a common problem for Omnibus boards. What's causing it remains a mistery so far. Possible causes:

1. Faulty acc/gyro chip on some batches
2. Gyro chip power supply noise issues
3. Heat issues from built-in LDO
4. Time jitter issues

All of those need to be considered.

skaman82, 2017-05-24:
@digitalentity I don't think that this is an omnibus specific problem. Having the same behavior with my SPRacingF3 EVO with external minimosd.

Kaffiflight, 2017-06-02:
Hello, new here. I signed in to join the conversation - I experienced the same Problem (Omnibus F3 user), currently INAV 1.7.0. I have 2 fixed wing with INAV right now. A slow AXN Floater for first testing and a S800 (small flying wing). At the AXN I noticed it on the horizon roll, on the S800 more likey at the elevator trimm. In both cases it seems to happen when I fly with some angle for more than just some seconds. Since the AXN is pretty tolerant with elevator, I noticed this problem while flying long and wide circles, returnig back to horizontal seemed to have shifted the line in the direction of the curve before.
With the Wing its more noticable with the elevator. you need some up elevator while flying slowly to keep it level. In horizon mode ot seems to "adapt" to this position and keeps it slightly nose up even with low or full throttle. But if I push it downwards for some time, to speed it up or lose height, it will adapt to that. And even while it should fly level (what it did maybe 2 minutes before), it tends to sligthly dive down more and has to be held with upelevator for a while. While doing so for some time, it returns back to this "new situation" and starts to keep the nose up again.
What I want to say with that:
1. It´s not only visible at the OSD artificial horizon but also in flightcaracteristics
2. It seems to happen while you hold an angle for a while and then drift back again.
3. It happens with 2 different (but same type) F3 Omnibus.
4. Right now I believe it´s not related to one of the sesors (one axis) but all. It´s only the AXN does not show so much effects on such small changes in elevator, but flies long and wide curves. the fyling wing is pretty fast rolling and doing circles, so I may have not noticed (and not tested) it yet about the roll drift.

Right now I don´t understand enough about it to know if it´s a bug or if it is related to some probelms I caused with wrong settings. Maybe I should also mention both F3 boards are mounted in different angles, none is mounted without being turned for 90 degrees on at least one axis.
Greetings from Germany

digitalentity, 2017-09-04:
Closed due to no further reports of the issue. I also can't reproduce the issue anymore.

closed 2017-09-04 by digitalentity