iNavFlight/inav #1446: blheli passthrough - flashing failed

eziosoft, 2017-03-22:
I think that might me important. I was setting up a quad for a friend, so I flashed latest inav and applied 5'' preset. And than I wanted to flash escs to newest blheli. And flashing for all escs failed leaving me with corrupted firmware on all escs. They still has bootloader intact so I tried many times and it always fails. Than I flashed betaflight and try to flash escs again and it worked on the first go. So something in inav is not working as it should.

even detection of escs wasn't "very good" usually blheli suite could find 3 of 4 escs. sometimes all 4. It was random which esc is not dedected.

someone else also have issues https://www.rcgroups.com/forums/showpost.php?p=37148383&postcount=12494

stronnag, 2017-03-22:
Works fine for me; didn't really need Imperial March or Games of Thrones tunes at startup ... but so what.

omg. Mrs Stronnag now wants to know why my machines are playing a passable imitation of Imperial March and Games of Thrones tunes. Sometime one wishes one had never bothered.

stronnag, 2017-03-22:
The above test was on my SP EVO, updating some previously flashed by me DYS 30A blheli ESCS.

I then connected some new Racestar 30A blheli_s (default factory flash) to my also new Omnibus F3AIO. Would not flash reliably using the Chrome app. Eventually flashed using blhei.exe + wine + SPRF3. Once updated (blheli 16.6) they will flash OK in either tool (chrome or .exe), on any board (SPRF3, SPEVO or Omnibus). 

So I'm not sure this is iNav problem as much as a board / esc / original esc firmware issue.

For what's worth. Now, if they'd only play the first few bars from Berlioz Te Deum before the choir comes in, it would awesome.

crashfrequently, 2017-03-22:
As part of an upgrade, I tried to flash the latest BLHeli into littlebee 30A esc;s using 1.6.1.
In my case #3 seems to have flashed OK but the other three failed and are now totally unresponsive.
It looks like their boot loaders have been lost/overwritten/corrupted.

stanislavdimov, 2017-03-25:
Do not work for me! Naze32 rev6 ESC kingkong Blheli 12A. 
I do not have problem when flash with cleanflight and betaflight, blhelisuit work perfect!
Message  : Initialization of serial 1 wire passthrough or 4way interface failed!

oleost, 2017-03-25:
@stanislavdimov https://github.com/iNavFlight/inav/wiki/Supported-boards#boards-based-on-f1-cpus

stanislavdimov, 2017-03-25:
Yes ...F1 board must do not flash with inav!  Cleanflight!

digitalentity, 2017-03-25:
@eziosoft what board and what ESCs was that?

crashfrequently, 2017-03-25:
As part of an upgrade, tried to flash using Omnibus F3, release 1.6.1 on Littlebee 30A esc.
Had the same experience as eziosoft. After numerous attempts, ended up with one OK but now can't communicate with the other three.
Previosly never had any problems using SPRF3 on RMDODO.

eziosoft, 2017-03-26:
@digitalentity spf3 (clone and original) and Racerstar MS Series 25A and 35A.

digitalentity, 2017-03-26:
I noticed thae problem with Racerstars as well. Many Racerstar ESCs I got from Banggood have dodgy bootloader that even bricked ESCs in a few cases. After flashing proper BLHeli bootloader via C2 interface all ESCs came to live and flashing via blheli passthrough never failed for them.

I'm not sure what's the issue with @crashfrequently's ESCs. Possibly something is different about the passthrough code. Does flashing through Betaflight work?

crashfrequently, 2017-03-26:
Betaflight didn't work, it looks like the bootloader is gone.
When I get a bit of time I'll try to recover them using C2 interface.

tonyintn, 2017-03-27:
I thought it was just me lol, mine did the same thing I am trying to recover it now.  (Naze, INAV 1.6.1, Racestar BLHeli_S 20A lite.  Flashed failed corrupted now dead.

tonyintn, 2017-03-27:
I have confirmed its definitely something in INAV.   I flashed betaflight on my Naze then I was able to flash and update my ESCs.

digitalentity, 2017-03-27:
@tonyintn I've made catch-ups with Betaflight regarding BLHeli passthrough code. If possible can you re-test with https://github.com/iNavFlight/inav/pull/1472

tonyintn, 2017-03-27:
Sure, let me make a custom build and I will try it on one of my other ESCs I have only updated that recovered on thus far.

tonyintn, 2017-03-28:
Gave me a bad write

digitalentity, 2017-03-28:
It's now exactly the same code as in Betaflight, no reason to expect different behavior. I'll test on my FCs.

tonyintn, 2017-03-28:
Let me try again may have forgot to point to #1472

tonyintn, 2017-03-28:
Same thing for me could very well be something I am doing not sure. Notice it recognize #1 but all the other are "u". 

https://www.dropbox.com/s/f0cwysf2sdr2kjz/Screen%20Shot%202017-03-27%20at%208.04.04%20PM.png?dl=0

https://www.dropbox.com/s/n1ycviu6eoi4p1i/Screen%20Shot%202017-03-27%20at%208.09.08%20PM.png?dl=0

tonyintn, 2017-03-28:
tried again must be something on my end I an not getting a clean build then I point to git checkout 8f13eec
get a the following:
%% dashboard.c
./src/main/io/dashboard.c: In function 'formatTrimDegrees':
./src/main/io/dashboard.c:557:2: warning: implicit declaration of function 'spri
ntf' [-Wimplicit-function-declaration]
sprintf(trim, "%d", trimValue);
  ^
./src/main/io/dashboard.c:557:2: warning: incompatible implicit declaration of b                         uilt-in function 'sprintf'
./src/main/io/dashboard.c:557:2: note: include '<stdio.h>' or provide a declarat                         ion of 'sprintf'

tigert, 2017-04-01:
Just for the record, had this too, omnibus f3 and MS 35A Racerstar (a single esc, I am building my S800 flying wing for inav).

Flashed OK with betaflight, 

I will see tomorrow if my build chain works to test #1472.

digitalentity, 2017-04-04:
See https://github.com/iNavFlight/inav/pull/1472 - tested on multiple boards and multiple ESCs without issues.

closed 2017-04-04 by digitalentity

darmach, 2017-07-13:
I had the same issue on latest iNav. Using SPF3 Deluxe, with original LittleBees 20A PRO and Gartt LT2206 2000kv motors. I'm not risking flashing blheli, as I'm unable to correctly check/discover all ESCs in the same time. Every time one of them fails, sometimes more than one.

Checked on second SPF3 board, the same. Then flashed cleanflight, and on both boards it works, tried few times.

tonyintn, 2017-07-13:
Yes the bug is still there; I have noticed this for sometime now there is still a difference with biheli between Inav and Cleanflight I just did not want to bother the dev about his as there is a workaround. 
Just backup your Inav setting then flash cleanflight and setup your ESCs then flash back to Inav and restore your settings.

darmach, 2017-07-13:
Yes, sure, the workaround do exist. Just been wandering if this shouldn't be open till fixed. Myself working in IT I always prefer to track issues. What's the approach in here?

tonyintn, 2017-07-13:
you would open a new issue as this thread is actually closed.

digitalentity, 2017-07-14:
@darmach please open a new issue for this. It's really odd since I don't have any issues with BLHeli passthrough on my builds :thinking: