iNavFlight/inav #1447: Can't ARM by switch. But can arm with stick combination.

DscoveryHD, 2017-03-22:
Can't ARM by switch. But can arm with stick combination. No GPS installed.
Model: flying wing.

[dump_1.6.1_23.03.17.txt](https://github.com/iNavFlight/inav/files/863007/dump_1.6.1_23.03.17.txt)

oleost, 2017-03-23:
Screenshot of your modes tab with armed turned switch turned on and off please.

DscoveryHD, 2017-03-23:
[image]
[image]
[image]

oleost, 2017-03-23:
Hmm, no idea why that happening for you, guess have to wait for someone else.

Just to be sure, you end points are showing 1000 in throtte and 1500us the rest right?

DscoveryHD, 2017-03-23:
While arming by switch:
Throttle 998 +-5
Yaw 1500 +-5 
Roll 1500 +-5
Pitch 1500 +-5
AUX3 2000 +-5
[image]

digitalentity, 2017-03-25:
I'll try reproducing this from my end.

digitalentity, 2017-03-27:
I can't reproduce the issue.

digitalentity, 2017-03-30:
Can't reproduce.

closed 2017-03-30 by digitalentity

DscoveryHD, 2017-04-02:
т.е. во всем виноваты мои руки? искать где накосячил, так?

digitalentity, 2017-04-03:
Более подробное описание что и как подключено может помочь. У нас уже был баг, который проявлялся только на конкретном оборудовании.