iNavFlight/inav #1455: GPS Coordinates not displaying using SmartPort Telemetry in OpenTX - SPRacingF3EVO

MRC3742, 2017-03-24:
After hours of trying different combinations of OpenTX firmware versions and iNav versions I am unable to show or log GPS Coordinates via SmartPort telemetry for my location using a Taranis X9D+. All other GPS related telemetry appears to be OK. It will not show GPS Copordinates in all stable versions of CleanFlight, iNav and BetaFlight up to and including v3.0.1. HOWEVER IT WORKS PROPERLY using BetaFlight on any firmware versions higher than 3.1.0.

Firmware versions listed in descriptions below

The screenshots shown below are using OpenTX v2.1.9 and iNav 1.6.1 and Although nothing appears on the telemetry screen (Pics 1 and 2) on the edit screen the longitude value shows correctly (Pic 3 upper row)
[image]          [image]           [image]

Using BetaFlight with firmware versions 3.1.0 and newer, the following appears correctly on telemetry screen.
[image]

Updating OpenTX to version 2.2 RC14 (N366) does change the display somewhat showing the latitude as all 0's and the longitude now shows on the telemetry screens but in the second location. (As it was shown
 in the first location slot on OpenTX v2.1.9)
[image]         [image]          [image]

Again using BetaFlight firmware versions 3.1.0 and newer, with OpenTX V2.2 RC14 the following appears correctly on telemetry screen. (the Lat/Lon are on opposite rows from earlier OpenTX v2.1.9 with latitude now appearing on the top row)
[image]

Hardware I am using: 
Taranis X9D+ with OpenTX 
SPRacing F3 EVO (Not a Clone) 
Ublox M8N GPS (Clone) on Uart3 with external HMC5883 Compass on I2C1
FrSky XR4-SB using PPM and SmartPort Telemetry on Uart2
I have also moved GPS and Smartport to different Uarts including SoftSerial with same results

Also GPS Coordinates DO SHOW PROPERLY in iNav configurator in the setup and GPS screens.

I have searched the forums and it was reported to be working form their home location but changing to a different area seems to make it stop working?

MRC3742, 2017-04-29:
Haven't been able to get GPS Latitude Coordinates to display on Taranis LCD. Also logging csv file GPS column is blank using smartport telemetry on X4R-SB.

It will work if I change RX to D8R-XP and iNav setup from Smartport to Frsky using hub telemetry on iNav 1.6.1.

I am able to compile my own hex and have been working in smartport.c but have not been able to get latitude to display. I believe gpsSol.llh.lat is not available here? Switching gpsSol.llh.lat with gpsSol.llh.lon in smartport.c causes the longitude coordinate to be displayed as latitude (My negative value is displayed as South instead of North as expected) but the longitude on the display (which is now set as gpsSol.llh.lat) is now displayed as 0's

As I have very limited knowledge to programming code, am I on the right track and working in the right file?

DzikuVx, 2017-06-09:
I can confirm GPS coordinates working with X4R receiver and OpenTX 2.2 on Taranis and INAV 1.7.
Closing this one

closed 2017-06-09 by DzikuVx