iNavFlight/inav #1458: "feature motor_stop" vs "RTH"

Schalonsus, 2017-03-25:
If feature "motor_stop" is enabled, it is possible to shut down motor when in RTH mode with throttle stick to zero.

Linjieqiang, 2017-03-26:
No.It will be dangerous.But in DJI product drone,you just only put yaw and pitch channel at 1000 , throttle and roll channel at 1000 ,the motor will shut down.

Schalonsus, 2017-03-26:
Yes it is a bug, thats why i report it!

digitalentity, 2017-03-26:
This is indeed dangerous, but to be honest I'm against overriging MOTOR_STOP by anything but failsafe.

MOTOR_STOP is intended to be a safety feature giving pilot the ability to quickly shutdown motors in case of the emergency.

oleost, 2017-03-26:
Yeah, I noticed this earlier on my airplane. But figured it was by design as motor_stop is intedend to stop motors at 0% throttle.

Schalonsus, 2017-03-26:
In passthrough yes, but in Angle/RTH/PosHold/AltHold/WP? 
I consider this more unsafe... But maybe we should just state this somewhere in the wiki so that people are aware if this "function"

oleost, 2017-03-26:
In angle mode its obvious its going to stop the motor.

But think for people that are not using a switch to arm. Those maybe choose motor_stop to have an easy way to cut throttle.

andre92, 2017-03-26:
Looks like this is intentional behaviour. Motor_stop stops the motors when throttle is lowered. It is like a kill switch. When using motor_stop, consider setting up an idle up switch on your transmitter so you don't kill the motors by accident. Or when switch arming, use a 3 position switch for arm + idle up. It doesn't use extra aux channels.

Dronek, 2017-03-27:
Actually I never considered motor_stop as a safety feature.
From my understanding currently only motor_stop allows for gliding with a folded prop AND full PI stabilization.
(Switch-)disarming is no option to glide a plane due to obvious reasons.

I support the idea of overriding motor_stop by failsafe, at least for fixed-wing.

digitalentity, 2017-03-27:
Failsafe always ignores motor_stop. If we are in failsafe and system requires motor to spin - it will spin.

andre92, 2017-03-27:
@Dronek Sure I agree, but for fixed wing vs multirotor, motor_stop has a different use. Multirotors don't glide too well so I would only use motor_stop on fixed wing. Motor_stop on multirotors can be dangerous and I prefer not to use it.

Dronek, 2017-03-30:
> You could consider it a safety feature on multirotors if you don't have an arm switch but want to stop the motors asap. Stick disarming without motor_stop still takes a few seconds at idle.

That's true. It seems that motor_stop has quite different applications and implications when it comes to multirotors vs. fixed-wings.

@all : would it make more sense to not use motor_stop on fixed wing and just set min-throttle to 1000?
This would prevent from "accidentally" cutting throttle in nav modes while still allowing to cut throttle in modes like angle, acro, pass-through etc.

Redshifft, 2017-03-30:
@Dronek  The system works, there has not been a need to tell people not to turn off their transmitters while flying, so after a couple of long walks they will figure this one out also :D

hali9, 2017-03-30:
I use always this feature and I agree with @digitalentity:
> MOTOR_STOP is intended to be a safety feature giving pilot the ability to quickly shutdown motors in case of the emergency.

MOTOR_STOP should always stop all motors if pilot have control. One and only one exception should be failsafe when pilot do not have a control.
It is sometimes the last resort to do no harm to anyone.

giacomo892, 2018-01-08:
is this still actual or has being solved?

fiam, 2018-01-08:
@giacomo892 Thanks for noticing it. Fixed in https://github.com/iNavFlight/inav/commit/0778619547fd33b62a2119b0433dcc1b2a517c0b

closed 2018-01-08 by fiam