iNavFlight/inav #1459: Possible failsafe bug

digitalentity, 2017-03-26:
Reported on RCG: https://www.rcgroups.com/forums/showpost.php?p=37182329&postcount=12611

Possible bug with failsafe locking user out completely unless in passthrough mode (fixed-wing).

oleost, 2017-03-26:
@digitalentity https://www.rcgroups.com/forums/showpost.php?p=37185149&postcount=12636

oleost, 2017-03-26:
[image]

https://github.com/iNavFlight/inav/blob/master/docs/Failsafe.md
[image]

digitalentity, 2017-03-27:
Pilot lockout is possible when failsafe disarms the machine - it then locks pilot out of control for 30 seconds (including PASSTHROUGH mode). Fix pending https://github.com/iNavFlight/inav/pull/1465

closed 2017-03-27 by digitalentity