iNavFlight/inav #1460: Extend RX_MSP to be RX_TELEMETRY

digitalentity, 2017-03-26:
Upcoming bi-directional MAVLink will have a possibility to pass control via uplink. To allow this type of control we can re-use existing RX_MSP infrastructure which is by design control via telemetry uplink.

fiam, 2018-02-18:
@digitalentity Is this one still relevant? As our RX projects have shown, RX_MSP is capable enough to be used by an RX and telemetry data can be obtained by using the already implemented MSP commands (battery, attitude, coordinates, etc..).

digitalentity, 2018-02-18:
@fiam it's a different issue. Probably poorly named 😄 

It's about allowing RC control via telemetry uplink (MSP, MAVLink etc).

RomanLut, 2022-06-17:
I suggest to close this as completed.
Mavlink RC control has been implemented recently https://github.com/iNavFlight/inav/pull/6738
For MSP, MSP_SET_RAW_RC command exists for a long time already.

closed 2022-06-17 by DzikuVx