iNavFlight/inav #1468: Remove 500hz gyro limit on F1 targets.

andre92, 2017-03-27:
For people who know what they are doing. The 500hz synced default is pretty safe.
When using Asynchronous mode on F1 targets, it is theoretically possible to set the gyro faster than 500Hz. I have succesfully done so, with gps. However, iNav has a hardcoded gyro limiter on F1 targets that limits the gyro to 500hz with gps enabled. I removed it and enabled asynchronous mode on a custom Naze build.
Could this limiter be removed from future versions so that we can fly F1 targets in asynchronous mode with GPS enabled?
my settings:
async mode all
gyro 1khz
pid 500hz
acc 500hz
attitude 250hz
i2c overclock on
I attached my custom 1.6.1 build. I removed gyro limiter, added async mode and notch filters, and cli task statistics. Only supports PPM RX and FrSky Telemetry.
output of tasks:
Task list         rate/hz  max/us  avg/us maxload avgload     total/ms
 0 -       SYSTEM       9       7             4    0.5%    0.5%         5
 1 -              PID     497     697     468   35.1%   23.7%     30840
 2 -          GYRO     980     345     237   34.3%   23.7%     28734
 3 -            ACC     496     194     183   10.1%    9.5%     11181
 4 -   ATTITUDE     245     344     329    8.9%    8.5%     10007
 5 -               RX      49     124     113    1.1%    1.0%       690
 6 -         SERIAL      99   74124      54  734.3%    1.0%       300
 7 -      BATTERY      49      23      10    0.6%    0.5%        66
 8 -        BEEPER      98      20       7    0.6%    0.5%        87
 9 -             GPS      24     457      97    1.5%    0.7%       301
10 -   COMPASS       9     144     138    0.6%    0.6%       170
11 -         BARO      43     144      97    1.1%    0.9%       502
12 -  TELEMETRY     228     116       6    3.1%    0.6%       166
Task check function             9       5                        31
Total (excluding SERIAL)                    97.5%   70.7%

[minimal_inav_1.6.1_NAZE.zip](https://github.com/iNavFlight/inav/files/872450/minimal_inav_1.6.1_NAZE.zip)

oleost, 2017-03-27:
Aren't async mode removed from F1 as well?

digitalentity, 2017-03-27:
That limit is imposed by the fact that F1 board will be incapable of maintaining 1kHz gyro AND GPS modes activated. However you can try actually flying AltHold + PosHold with your config and post a BB log so we can confirm that 1kHz on F1 is viable.

However I don't think we should try to keep F1 up to date for much longer. They are obviously EOL.

andre92, 2017-03-27:
I was just playing with custom builds and ran into this. However, 1500 looptime (666Hz) is still possible in synced mode with i2c_overclock on and gyro_lpf = 256Hz.

andre92, 2017-03-28:
Here is a blackbox with 1k gyro, 500hz pid, 500hz acc, 250hz attitude and i2c_overclock on Naze. Logging rate was 100%, so I only got a minute of log. It contains poshold and RTH and no glitches have occured during this flight. Which logging rate would you suggest for my next log?
[blackbox_log_2017-03-28_211904.TXT](https://github.com/iNavFlight/inav/files/876968/blackbox_log_2017-03-28_211904.TXT)

fiam, 2017-09-09:
I guess we can close this one now that F1 targets are out

digitalentity, 2017-09-10:
Targets based on F1 chip are no longer supported. Closing.

closed 2017-09-10 by digitalentity