iNavFlight/inav #1473: Add possibility to disable failsafe entirely

digitalentity, 2017-03-28:
I have received a request to allow failsafe to be disabled entirely - no safeguards on the channel values, no fallback values in case of signal loss. Just allow FC to always trust the data that's coming from the receiver regardless of it's quality and reported RC link state.

What do you think?

Linjieqiang, 2017-03-28:
It make fc be unsafety.I wonder why the idea grows.In my opinion,the fc will provide essential protection to our plane or copter Such as RTL or Landing safely.

digitalentity, 2017-03-28:
@AlouetteIII you are not thinking about bigger picture. Failsafe is not about RC link loss exclusively. Successfully handling RC link loss requires knowledge about system state as a whole. Leaving FC out of the loop and assuming that aircraft is always able to execute any command is potentially very unsafe.

Imagine scenario where you want to activate RTH in case you loose control - common scenario. But imagine that by an unfortunate coincidense there is no GPS coverage at the moment (you are in the area of high RF interference).

**Configuration 1: No failsafe on the FC, receiver is set to activate RTH mode:**
In this case RTH will not activate at all. FC will do nothing. Most likely result - a fly away.

**Configuration 2: You configure failsafe on the FC to do RTH:**
In this case FC will see that it is in RC loss and GPS loss and execute a different procedure - emergency landing considerably increasing chance of successfully recovering the bird.

Redshifft, 2017-03-28:
I would stick the original Multiwii/baseflight code back in and add a Failsafe mode = it always worked even picked up serial RX FS flags

AlouetteIII, 2017-03-28:
@DE - Most of my pilot friends who reported they lost a plane/drone from flyaway or blow away and searched for it for days but most never found these (you included) as if already far away and then lands it's a low probability of happy finding ; However in fact Many/majority reported numerous safe hardware based Rx RTL situations ; and 100% amongst those pilots who use / understand Receiver-Hardware failsafe & Tested it prior. Software-Failsafe "alternative" is less reliable (see forums) and is something that could be useful one day to deal with redundancy on a whole different level/logic - such as flying by 2ndry effects, RTL by Dead Reckoning logic DR (time and heading nav) - that would be nice ; but it must never compete with hardware LVL 1 protection which has proven itself reliable already.

digitalentity, 2017-03-28:
I suggest doing like this:

Software failsafe in INAV will override what RX is telling it anyhow (in all modes except NO ACTION).
This means I can safely take out signal loss behavior logic from RC RX code. This means that values from RC will always make it to the firmware (and show in RC receiver tab), but in some cases they will be ignored by software-based failsafe. 

If one so desires he configure **failsafe_action = NONE**. This will effectively disable failsafe logic. FC will detect signal loss and start beeping S-O-S but apart from that no action will be done.

Does this sound reasonable?

Redshifft, 2017-03-28:
Yes DE do it it does not work reliably anyway

digitalentity, 2017-03-28:
Hacked some code together. Behaves pretty much like I described now:

Receiver unplugged:
RC channels freeze, red parachute lights up

TX off, S.Bus RX configured to "hold"
RC channels freeze, red parachute lights up

TX off, S.Bus RX configured to "set"
RC channels move to pre-set values, red parachute lights up

Tests with PWM/PPM receiver pending.

digitalentity, 2017-03-28:
@AlouetteIII you should fly in our hills. GPS coverage is sporadic - too much RF interference from nearby military radars. 

If the plane will continue it's flight - it's gone in the woods where there is no chance of recovery. If it will land - there is a slight chance of recovering it (especially if it is equipped with long-range autonomus beacon). Unfortunately my plane won't - lession learned.

While I don't appreciate leaving FC out of the loop in case of failsafe event - I'll make it works like people want it to and let everybody do it the way they like it.

Redshifft, 2017-03-28:
@digitalentity what was it based on ? did you try old Baseflight?

@AlouetteIII DE can fix that no problem but lets deal with the RClink fail first ;)

digitalentity, 2017-03-28:
Baseflight code is a mess. I'm taking out signal override from the current code which is at least readable. Basically same behavior as in Baseflight.

Redshifft, 2017-03-28:
Nice :D

AlouetteIII, 2017-03-28:
I like the new proposal 4 posts above - especially if i can test this in the GUI - so when i turn Software-Failsafe = OFF ;    its 100% off ; it's a Duty of Care thing for each individual.  Actually my quad was swamped by a cell tower once (no GPS in that racer) - my FrSky D4RII was pre-coded for a descent low throttle stabilised condition. it worked.   The Rc was jammed but the Rx failsafe saved it because i had taken the time to pre-code my throttle/modes in ;-   Don't like the way you use the word FAILSAFE - it's software failsafe - the majority of pilot have paid for and have excellent HArdware-Failsafe in their Rx's ;     For future features Let's calculate and store wind vector -  DR nav as a backup to GPS ; combined with inertial for the GPS just failed situation ; thats what i call smart failsafe.

oleost, 2017-03-28:
Remember an wire can also get loose between RX and FC, and it's not uncommon.

AlouetteIII, 2017-03-28:
Solder your SBus wires directly to the FC

hali9, 2017-03-28:
I use Flysky TX/RX, and I have following problem.
In Flysky are two option when RX lost signal: kep last channel value or set up value. But this can be set only for 6 first channel.
Previosly I use old FS-IA6B which have 6 channel PPM, so I set up failsafe in RX and works ok.
Currently I have new FS-IA6B which have 8 channel PPM, so 6 first channel are set, but 7 and 8 no.
If I turn off TX, 7 and 8 channel can kep last value or can have random value.
If random value is beetween accept value (rx_min_usec, rx_max_usec) then FcFailsafe is not fired.
If it outside (rx_min_usec, rx_max_usec) then FcFailsafe is fired.
More: I can use IBUS which have 10 channel and I have 6 first channel set and 7,8,9,10 not.
I want use at first my RX failsafe (because I don't disable it).
And FC failsafe should be use only when wire are cut off between FC and RX.
For this to work, channels above 6 should be ignored.
The failsafe should only include channels 1-6.

sppnk, 2017-03-28:
Long range RF beacon.. :-) I use for now openlrsng and I have recovered several times my birds with the RDF method. It works.

Now I use openlrsng telemetry with LTM and it is much easier (EZGUI map), but if this fails I have still the RF beacon.

Recently I am testing LoRa stuff for independent beacon. It seems to be the best solution, but adding a GPS it weights, so only for big birds.

And... finally I have purchased a RF-V16 tracker.

I am paranoid. :-)

sppnk, 2017-03-28:
Regarding to disable failsafe. I dont see the point of that.

Failsafe should be easy to setup, bomb proof and not much code and parameters involved. Just select RTH, land or kill all. IMHO.

Redshifft, 2017-03-28:
@hali9  you only need 1 channel to call failsafe, sure you can find a spare with FS preset ability.

Redshifft, 2017-03-28:
@sppnk  what's with "Disable" it is just being changed because it does not currently work

hali9, 2017-03-28:
@Redshifft I know that one channel can call failsafe. 
The problem is call failsafe if any of channel 1-6 is wrong and not call failsafe by channel 6-10.
Channel 6-10 may be any value and it not is reason to call failsafe.
This can not be set in any way.
in file rx.c: 
function static void detectAndApplySignalLossBehaviour(void)
 call bool validPulse = isPulseValid(sample); for all channels.

Redshifft, 2017-03-28:
@hali9 Sorry i did not quite understand that, but I'm sure any channel calling the "FS mode" will not be overridden by other non settable channels jumping around.

oleost, 2017-03-28:
#1475

closed 2017-04-11 by digitalentity