iNavFlight/inav #1475: Figure out how we want Failsafe to function.

oleost, 2017-03-28:
Hi.

There have been some modifications to Failsafe, and there have been various issues.

Before actually modifying it would be beneficial to know where we are headed and what we want.

I started drawing and diagram. Feel free to comment and come with improvements.

[image]

@hydra @borisbstyle maybe your interested in this as well.

sppnk, 2017-03-28:
Great idea @oleost , FAILSAFE understanding and issues related to it were growing lately like a monster...

oleost, 2017-03-28:
[image]

digitalentity, 2017-03-28:
It's somewhat similar to what we already have, but we now have the roadmap, good job @oleost

hali9, 2017-03-29:
In box pulse detection.
Can be added which channels are to be checked ?
For example only 1-6. https://github.com/iNavFlight/inav/issues/1473#issuecomment-289687490
I think about CLI command like failsafe_watch_channel uint16 default 0xFFFF as bitwise combination, for example:
0000 0000 0000 0100 -> failsafe only when channel 3 (throttle) is out of (rx_min_usec, rx_max_usec).

And if failsafe_watch_channel[n] is 0 then set:
if channel[n] < rx_min_usec then channel[n] = rx_min_usec,
if channel[n] > rx_max_usec then channel[n] = rx_max_usec,

oleost, 2017-03-29:
@digitalentity Almost yes, but not 100%.

We have some quirk's and some timers which one could maybe ask ourselves if we really want.

hydra, 2017-03-29:
pulse detection loss must include recevier wire cut/broken/unplugged.

oleost, 2017-03-29:
@hydra  Yes, is it not that no pulse detection are mainly for?

The preferred method for telling the FC that link between RX and pilot is lost is digital failsafe flag or jump to preset position that activates "Failsafe" flight mode

Redshifft, 2017-03-29:
@hydra Oi, what do you know about it? you already buggered up your own one ...Lol

bzdang, 2017-03-29:
@hydra Thank-you Dominic for your (vast) contributions, past, present (& future).

Antonio76m, 2017-05-06:
hi guys, any update on this feature? would be great to have a voltage failsafe :)

krzysztofmatula, 2017-05-08:
If the voltage failsafe is to be supported, the 'vbat' would need to be averaged within relatively long time windows. I observe the voltage measurements tend to fluctuate heavily even during stable flight. 

See: https://github.com/iNavFlight/inav/files/980996/blackbox_log_2017-05-06_154324.TXT
The 'vbat' fluctuates for at least 2V all the time.

I'm using SPF3 here. Is it the same for other boards?

teckel12, 2017-09-28:
I can't imagine a battery voltage failsafe working as intended.  It would either trigger too early due to voltage sag or trigger too late where RTH takes far longer than battery will allow.  My opinion is that only mAh remaining would be an effective way of triggering a failsafe.  I do this via a LuaScript currently.

closed 2018-05-27 by stale[bot]