iNavFlight/inav #1480: Constant short beep every ~500ms

krzysztofmatula, 2017-03-29:
I'm trying inav 1.6.1 on SPRACINGF3 and what I'm getting is that the FC keeps producing a short beep every ~500ms. It persist before and after arming. Together with the beep the red LED blinks.

Any idea what can it be?

In the configurator everything seems to be fine, all pre-arming checks are green, the hardware is healthy, receiver is working, GPS has the fix.

I have a 3-positional switch assigned to change the rate-profile and I observe that the beep (and LED blink) is gone after selecting profile #2, while it gets back when selecting profile #1 again.

In the 1st profile I have more or less a default setup for 10inch MR.
Here's my diff:

```
# diff

# version
# INAV/SPRACINGF3 1.6.1 Mar 28 2017 / 18:51:24 (a3c93f9)

# resources

# mixer

# servo

# servo mix

# feature
feature SOFTSERIAL
feature GPS
feature TELEMETRY
feature RSSI_ADC
feature LED_STRIP
feature PWM_OUTPUT_ENABLE

# beeper
beeper -ON_USB

# map

# serial
serial 0 1 115200 57600 0 115200
serial 1 2 115200 57600 0 115200
serial 2 16 57600 57600 57600 115200
serial 31 1 9600 38400 0 115200

# led
led 0 15,15:S:C:2
led 1 14,15:S:CW:2
led 2 9,15:S:CW:2
led 3 8,15:S:CW:2
led 4 7,15:S:CW:2
led 5 6,15:S:CW:2
led 6 1,15:S:CW:2
led 7 0,15:S:C:2
led 8 11,0:N:C:1
led 9 10,0:N:CW:1
led 10 9,0:N:CW:1
led 11 8,0:N:CW:1
led 12 7,0:N:CW:1
led 13 6,0:N:CW:1
led 14 5,0:N:CW:1
led 15 4,0:N:C:1

# color

# mode_color

# aux
aux 0 2 1 900 1200
aux 1 4 1 900 1075
aux 2 4 1 1250 1350
aux 3 3 0 1100 1250
aux 4 3 0 1350 1475
aux 5 3 0 1600 1750
aux 6 3 0 1900 2100
aux 7 9 0 1250 1500
aux 8 8 0 1500 2100
aux 9 11 1 1500 2100
aux 10 12 2 1700 2100
aux 11 18 1 1500 2100

# adjrange
adjrange 11 2 2 900 2100 12 2

# rxrange

# master
set looptime = 1000
set gyro_sync = ON
set gyro_sync_denom = 1
set gyro_hardware_lpf = 188HZ
set gyro_lpf_hz = 70
set gyro_notch1_hz = 170
set gyro_notch1_cutoff = 125
set gyro_notch2_hz = 85
set gyro_notch2_cutoff = 43
set acc_hardware = MPU6050
set acczero_x = 48
set acczero_y = -31
set acczero_z = 12
set accgain_x = 4082
set accgain_y = 4078
set accgain_z = 4078
set align_mag = CW270FLIP
set mag_hardware = HMC5883
set mag_declination = 450
set magzero_x = 116
set magzero_y = -54
set magzero_z = -17
set mag_hold_rate_limit = 30
set baro_hardware = MS5611
set blackbox_rate_denom = 4
set motor_pwm_rate = 1000
set motor_pwm_protocol = ONESHOT125
set failsafe_procedure = RTH
set nav_extra_arming_safety = OFF
set nav_mc_hover_thr = 1300
set async_mode = GYRO

# profile
profile 1

set mc_p_pitch = 75
set mc_d_pitch = 18
set mc_p_roll = 75
set mc_d_roll = 18
set dterm_notch_hz = 125
set dterm_notch_cutoff = 90
set rate_accel_limit_roll_pitch = 400
set rate_accel_limit_yaw = 180
set roll_rate = 40
set pitch_rate = 40

# 
```

krzysztofmatula, 2017-03-30:
Update:
this line causes the issue:
```
adjrange 11 2 2 900 2100 12 2
```
I was using this to change "rate profile" with 3-positional switch.
It seems we no longer have rate profiles in inav, right?
So, what this setting is actually doing?

digitalentity, 2017-04-04:
@krzysztofmatula no, we don't have rate profiles anymore. ID for rate profile change was kept for compatibility and that line actually does nothing apart from beeping.

krzysztofmatula, 2017-04-04:
Ok, so it's clear now, thanks :)

closed 2017-04-04 by krzysztofmatula