iNavFlight/inav #1482: I can't get telemetry with my frsky x8r receiver after I follow the docs done all steps, anyone had same problem as me?

hzc916487783, 2017-03-30:
I am so confused

andre92, 2017-03-31:
Issues and Support.

Please remember the issue tracker on github is not for user support. Please also do not email developers directly for support. Instead please use IRC or the forums first, then if the problem is confirmed create an issue that details how to repeat the problem so it can be investigated.

Issues created without steps to repeat are likely to be closed. E-mail requests for support will go un-answered; All support needs to be public so that other people can read the problems and solutions.

Remember that issues that are due to mis-configuration, wiring or failure to read documentation just takes time away from the developers and can often be solved without developer interaction by other users.

Please search for existing issues before creating new ones.
Developers

Please refer to the development section in the docs/development folder.

hzc916487783, 2017-03-31:
`# dump

# version
# INAV/SPRACINGF3 1.6.1 Mar  9 2017 / 00:47:57 (6c878ae)

# resources

# mixer
mixer AIRPLANE

mmix reset

# servo
servo 0 1000 2000 1500 90 90 100 -1
servo 1 1000 2000 1500 90 90 100 -1
servo 2 1000 2000 1500 90 90 -100 -1
servo 3 1000 2000 1500 90 90 -100 -1
servo 4 1000 2000 1500 90 90 -100 -1
servo 5 1000 2000 1500 90 90 -100 -1
servo 6 1000 2000 1500 90 90 100 -1
servo 7 1000 2000 1500 90 90 100 -1

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature VBAT
feature RX_SERIAL
feature SOFTSERIAL
feature GPS
feature TELEMETRY
feature RSSI_ADC
feature PWM_OUTPUT_ENABLE
feature OSD

# beeper
beeper GYRO_CALIBRATED
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
[... 399 lines cut ...]
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 20
set pitch_rate = 15
set yaw_rate = 9
set tpa_rate = 0
set tpa_breakpoint = 1500

#

hzc916487783, 2017-03-31:
I am sorry, english is not my native language ,I recheck many times around setting and wiring, but it still not woking, when I use clean flight, it works fine.

andre92, 2017-03-31:
Thanks for your config dump :) I found an error in your port configuration. You have set the baudrate of your telemetry port to 9600, which is for old FrSky telemetry on D-series receivers. For X-series receivers with a smartport it should be 57600 or AUTO.

hzc916487783, 2017-04-01:
`# dump

# version
# INAV/SPRACINGF3 1.6.1 Mar  9 2017 / 00:47:57 (6c878ae)

# resources

# mixer
mixer AIRPLANE

mmix reset

# servo
servo 0 1000 2000 1500 90 90 100 -1
servo 1 1000 2000 1500 90 90 100 -1
servo 2 1000 2000 1500 90 90 -100 -1
servo 3 1000 2000 1500 90 90 -100 -1
servo 4 1000 2000 1500 90 90 -100 -1
servo 5 1000 2000 1500 90 90 -100 -1
servo 6 1000 2000 1500 90 90 100 -1
servo 7 1000 2000 1500 90 90 100 -1

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature VBAT
feature RX_SERIAL
feature SOFTSERIAL
feature GPS
feature TELEMETRY
feature RSSI_ADC
feature PWM_OUTPUT_ENABLE
feature OSD

# beeper
beeper GYRO_CALIBRATED
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
[... 400 lines cut ...]
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 20
set pitch_rate = 15
set yaw_rate = 9
set tpa_rate = 0
set tpa_breakpoint = 1500

# 
`

hzc916487783, 2017-04-01:
thanks for helping and attention, I follow your instructions, insert that line, changed the baudrate, unfortunately  its still not working, telemetry didn't appear on my screen,   I already tried  different baudrate of telemetry port on softserial previously, when check the tasks, it seem like works fine, but I just didn't get telemetry.
`# tasks
Task list         rate/hz  max/us  avg/us maxload avgload     total/ms
 0 -       SYSTEM       9      11       4    0.5%    0.5%        25
 1 -          PID     495     738     683   37.0%   34.3%    215075
 5 -           RX     117     146     138    2.2%    2.1%      8807
 6 -       SERIAL      99   961851       6 9522.8%    0.5%      1366
 7 -      BATTERY      49   40257       6  197.7%    0.5%       235
 8 -       BEEPER      99      24       8    0.7%    0.5%       502
 9 -          GPS      24     609     120    1.9%    0.7%      1864
10 -      COMPASS       9     248     233    0.7%    0.7%      1469
11 -         BARO      99     265     245    3.1%    2.9%     15539
14 -    TELEMETRY     249      20       8    0.9%    0.6%      1335
17 -          CMS      58      15       6    0.5%    0.5%       208
Task check function            18      10                       676
Total (excluding SERIAL)                   245.2%   43.3%

# `

andre92, 2017-04-01:
I found one more setting worth trying: 
smartport_uart_unidir = OFF
try setting it ON, maybe that does the job. 
It has something to do with connecting both wires or TX only.

hzc916487783, 2017-04-01:
thanks very much  for helping, but  unfortunately  I tried this before, its still not working,  do you use frsky x8r receiver?  I don't know if there is anyone had same problem, I'm just so confused. I read the docs and check the setting  very carefully, maybe this is a bug, I don't had a clue~

puspis, 2017-04-07:
Hi. I show you my setup. SPRacingF3 + FRSky X8R (smartport & Serial Bus) + GPS.
I share UART1 with MSP and SmartPort. 
UART2 linked to GPS (38400 is a good value)
UART3 connected to SBUS on X8R (SerialRX)
And SoftSerial1 may be connected to OSD or 3DR telemetry.
See attached schematics.

Regards from Spain.
[image]

hzc916487783, 2017-04-09:
thank you very much for helping, I'm ready to  use Ominibus for iNav, it has integrated OSD, I abandon F3, I just can't figure it out, anyway thank you very much for attention,  best wishes for you!

DzikuVx, 2017-07-04:
OK, since topic appears to be dead and I can confirm X8R SmartPort telemetry working in latest releases with various connection scenarios, I'm closing this one. Thank you

closed 2017-07-04 by DzikuVx